InfiniTimeOrg/InfiniTime #363: Translation (i18n) support

christianfl, 2021-05-17:
As InfiniTime is only translated in English so far I thought it would be nice to get i18n support. This would improve UX and could make it easier to get new users in the future.

I found this but not sure wether this is suitable: https://github.com/lvgl/lv_i18n

Is it somehow realistic to use this or any other i18n support possibility keeping the limited resources of the PineTime in mind? Selecting languages could either be automatically by the Companian app (phone system language) or by a "Language" pane in the PineTime. I have no real C++ experience but after having some working code example I could implement the strings/translations and could provide English and German translations.

P.S. Thanks for InfiniTime to all contributors! Just got two sealed PineTimes for a friend and myself today and looking forward to wether or how I can contribute in the future.

Raupinger, 2021-05-18:
Lol, I thought of this yesterday and wanted to open an issue once I get to it. Anyways, here are my ideas and opinions:
I thought about this too and already looked at the lvgl implementation as well. I would prefer to have the language set at compile time to save space. My idea was to have a Header file that defines every string that needs to be localized and you just change the language by giving the compiler a different file. This approach would also fit nicely with needing to create different builds support umlaute or accents. It does have the drawback of making writing code a bit harder since you can't just put strings in there anymore. With the lvgl method you just call a method with the string and it handles everything. Ideally we would have that but with a compiler directive instead of a runtime function.

christianfl, 2021-05-18:
Hehe, nice coincidence 😃

I can totally understand the save spacing argument. However, couldn't we somehow get the best of both worlds? Just compiling with different flags means there would be independent builds for each and every language? This would sometime generate a long list of new builds and a user would have to flash it to change it...

I thought maybe it could be possible to use lv_i18n or some other gettext implementation and just push and apply the user language file via Companian app? Don't know how this could be done though. Using JSON or YAML for each translation instead of directly using C/header files has the advantage that its easier to translate by non-devs and these files are greatly supported by translation services such as Weblate or Transifex (if that's an option in the future). They get compiled to C anyway.

MFAshby, 2021-06-06:
I've made a start on this using lv_i18n for now. This was very easy to integrate into the InfiniTime build!

It should be possible to roughly calculate the application size increase from adding this additional translations. If it's too big, then it looks possible to tweak the build system to include only 1 set of translations in a firmware. I think this means we can use this existing tool without having to invent something new.

I think having a separate update mechanism for just the translations might be too complex for now. Companion apps (e.g. siglo) already provide some convenience UI for selecting the version you want to install, I think it could be relatively easy to add UI for flashing a particular language firmware.

JF002, 2021-06-10:
Once we integrate the support for the external memory, the size of the translation will be less an issue : we will be able to store translation files into the external memory instead of the internal one.

Generating multiple firmware with only one integrated language is much simple on the code level and more memory efficient, but we'll have to generate multiple bin files (I don't know how many languages we'll support).
Supporting languages as external resources is more flexible, but more complex and need the external flash integration.

MFAshby, 2021-06-24:
I've made a start on this anyway without the external memory https://github.com/JF002/InfiniTime/pull/417

At the moment this builds all the translations into the code, but later maybe it could be adapted to store on external memory. In the meantime, people could start contributing translations and building firmwares in their own language.

brainrom, 2021-07-07:
Hi there. I'm interested in i18n too, Russian language. @MFAshby work looks very well, but of course support for SPI flash is needed. I'll try to help with that, when I'll setup my dev env.  But I think we must leave an option to build the firmware with "hardcoded" translation by header with `#define _("example") "example"`-like macros.

LF-fr, 2022-02-19:
Hi,

Is this feature request still active ?

Avamander, 2022-02-19:
@LF-fr unless an issue is closed, it is waiting for its time.

aha999, 2022-04-03:
It would be great if there was a convenient way for general public to contribute translations if there aren't any already.

I can recommend [Weblate](https://weblate.org/), as it's popular, the platform itself is open source (compared to others), can be self hostable, and for open source projects like yours, they host translations for free. Weblate automatically merges translations and if you want you can have different widgets shown in github readme, that displays how much percentage each applied language is translated to in real-time.  It's much more simple for people to translate this way and it's free + less work needed for you.

- [set up your libre project](https://hosted.weblate.org/hosting/)

cquike, 2022-07-15:
Is there any progress on the support for languages other than English? I could contribute a Spanish translation.

cybuzuma, 2022-11-25:
I restarted internationalization in #1458. I found not much additional space is needed for translation, so adding some more languages should be possible.

aha999, 2022-11-26:
@cybuzuma if translation files took a lot of space, they could be installed manually, similar to new watchfaces (casio...)