InfiniTimeOrg/InfiniTime #365: Proposal: Allow multiple times / timezones

km6thj, 2021-05-18:
The core points of this proposal:

- Add a few time zones
- Optionally display a second time zone's time on a watch face

This is particularly useful when communicating often with people in other time zones, or when one needs
to see UTC often.

FoxAviation, 2021-06-09:
I'd be very happy if I were able to display UTC times, which is very important in some fields like aviation, maritime and, I'm sure, many others.

km6thj, 2021-11-03:
Really useful for amateur radio too.

geekbozu, 2021-11-03:
I feel like a lot of this should/could fall under 2 settings
One being a third time format
```
12-hour
24-hour
UTC
```
The second being watch-face specific information. Infinitime would need a way to get/set the time zone, but the display of it will likely be dependent on the watch face outside of the above representation options.

Avamander, 2021-11-03:
I do not want the complexity, space, time or implementation wise, that is dealing with timezones. At best I'd allow an optional setting for an offset clock.

geekbozu, 2021-11-03:
> 
> 
> I do not want the complexity, space, time or implementation wise, that is dealing with timezones. At best I'd allow an optional setting for an offset clock.
> 
> Time format is a separate setting, that's fine.

Yeah I think this is what I was poorly saying. I only want an offset from UTC stored in InfiniTime nothing more. Anything more is a waste of our effort. Updating it/maintaining it is the User/Companion apps problem. We should be a dumb smart watch that just shows the time we are given....eheheh

calhix, 2022-11-25:
Just wanted to mention I previously opened #1168 and didn't see this issue was open when doing a search since I think I only searched for world clock or different iterations of it. Mine is more about a separate app but being able to also display on a watch face is great also. I would rather swipe left to a world clock app from the watch face though.