InfiniTimeOrg/InfiniTime #366: Alarm

Raupinger, 2021-05-18:
Now that the timer is basically done its time for the last piece of the triforce, the (un)holy time tool: 
### The Alarm.
Beyond just putting in the work to build it, I see two main challenges. 
1. Logic
The timer could simply rely on system timers. Those work for longer periods and it would be nice to have alarms be persistent in the future. This means we need a daemon regularly checking if an alarm is due. That in turn brings the challenge of not unnecessarily wasting CPU time. I've looked around the docs and think that an endless loop calling vTaskDelay() for one minute would be a decent way to do this. This limits the resolution of timers to a minute, but I don't think that's a big deal.
2. UI
In contrast to the stopwatch and timer, a fully featured alarm system cant really be built using one screen. Here are some ideas of what it could look like (Disclaimer: example values may or may not represent values possible in reality :)
[image]

Raupinger, 2021-05-18:
also, we might reconsider putting timer and stopwatch together behind one icon. Having half of the first app list screen be taken up by various clock-like symbols would be kinda weird

Riksu9000, 2021-08-08:
I'm pretty sure the FreeRTOS timers can run for 48 days (2^32 / 1024Hz right?)

How would swiping from the list of alarms to the time setting screen work? How does it know which alarm to show on screen? Also there would have to be some indication that we can swipe so we'd know there's a hidden screen.

>also, we might reconsider putting timer and stopwatch together behind one icon. Having half of the first app list screen be taken up by various clock-like symbols would be kinda weird

If changing modes was done with swiping, again we would need some indication. If changing modes was done with buttons, it would take a lot of screen space, and at that point we might as well just keep them separate IMO. I'd rather have many simple apps than one complicated app.

Here's my proposal.

When we start the app, we get a list of four alarms. I suspect four will be enough. The dots represent days when the alarm is enabled. We would probably need to have some way to tell if the week should begin on Sunday.

If we click any of the alarms, we get to the second screen, where we can adjust the alarm. By then clicking set days, we get to the third screen.

Maybe if all days are deselected we can assume the alarm should run just once. Instead of the dots it could say "Once" to make it clearer.

[image]

Avamander, 2021-08-08:
Here's how Pebble's design team did it: https://youtu.be/5ulVS28MCNQ

(Might take a bit to process before it's visible)

adocampo, 2021-09-11:
Yes, pebble alarm was just perfect, imo. Not only it was super easy to set up with "weekdays", "weekends" and "once" presets, but besides, you were able to set it to a custom day set for example "mon, wed, fri and sun".
Besides, it had the snooze function, so when the alarm triggered, you could tap on that button and let the alarm trigger back in 5-60 minutes.
The vibration of the alarm was also quite cool: the typical military bugle melody to wake up the soldiers in the morning, very distinctive from a notification.

escoand, 2021-09-11:
@Riksu9000 I like your mock, looks similar to my Samsung Android alarm:

[image]

As you can see there are the first letters of the week days, so everybody knows whats meant. The first is a special weekday alarm which reminds you to go to bed.

[image]

Maybe this is something for the next steps but too special for now.

[image]

As you can see here are also only the first letters. Instead of the three switches there could be shortcuts for weekdays, weekends and so on.

mruss77, 2021-09-11:
Hey folks - I created a pretty simple alarm app (one screen, only one alarm supported). I like what's being proposed here, but in the meantime I needed something basic to get me up in the mornings.

Maybe some of that code can be used to build what you're proposing. Happy to help out.

GaryM0101, 2021-09-26:
The current setup on the i(information) screen gives a very nice "countdown" feature! I've only seen this in higher end digitals like the Yes watch. Having a countdown timer expands the alarm for use by airline pilots, NASA and others who want to know how much time is left on a task or experiment. So being able to choose particular a day of the week say 7 days in advance would be a good feature for this alarm.

mruss77, 2021-09-26:
I did some testing with different LVGL widgets to try make it easier to set the alarm time. It seems like a good UI for setting hours:minutes or minutes:seconds could be used for the alarm, the timer, and setting the time locally on the watch (#472).

This one shows setting hours and minutes with the slider in 12-hour format. If the watch is set for 24-hour format, we can just hide the AM/PM button and have hours go from 0-23.

https://www.youtube.com/watch?v=qc-q6q0IXjw

It's not bad, but as you can see it takes a pretty steady hand to precisely set the minutes.

In this one, I've sort of re-created the Android UI for setting times with the arc widget. In this case you would step through setting the hours and then the minutes on separate screens:

https://www.youtube.com/watch?v=iLrSRRIFZl0

The main problem with this one is that there's a "hard stop" at the 12:00 point. You can see in the video where the minutes are at 55 if you want to go back to the upper right side of the circle, you have to go the long way around.

It also has the same problem where it's not that easy to set a precise value. I suppose we could add +/- buttons for fine tuning.

I also experimented with the roller, but it didn't seem to behave well with 60 options for setting the minutes.

Another option is how @Riksu9000 handles this in the timer by allowing the button to be held down to repeatedly increase/decrease the numbers without having to press the buttons over and over.

Just wanted to throw this out there and see what people think about these different UIs. I guess as a last resort we could create a custom widget.

Riksu9000, 2021-09-27:
I think the LVGL slider widget can be used to create a kind of "trackpad", and the sensitivity could be set to make it easier to set precise values.

This could be accomplished by saving the slider value at `LV_EVENT_PRESSED`, then on `LV_EVENT_PRESSING`, add the difference between the old and the new value to the time. The knob part should be hidden, as it wouldn't act like a slider anymore.

There's also a separate issue for improving time setting #671

mruss77, 2021-09-27:
OK, I saw your mention of the trackpad in #671 but didn't fully understand it. That does sound like a good idea.

adocampo, 2021-09-27:
I was thinking on the dropdown shown on that application, which honestly, I'm still figuring if it's a metronome or what.
[image]

But it seems a quick way to pick a number, at least for the hours or finetunning the minutes

closed 2021-10-02 by Avamander