InfiniTimeOrg/InfiniTime #367: stop call notification when the call is handled by the phone

Raupinger, 2021-05-18:


raananelefant, 2021-09-30:
Would be nice for alarm clock as well :)

pbryan, 2022-01-21:
The PineTime will continue to vibrate indefinitely, even if the call is not answered by the phone. (e.g. the caller hangs up, the call goes to voicemail). This could probably be solved with a simple alert timeout?

JF002, 2022-01-22:
The current API does not provide anyway for the companion app to notify that the call was answered, ended, refused,... so yeah, the simple workaround will probably be to stop the alarm/notification after a few seconds. 
Another solution would be to implement an API that extends the current one to provide that functionality.

NCC1701M, 2022-03-23:
What's the progress on this? This is a pretty annoying bug.

Please fix as soon as possible

Riksu9000, 2022-03-23:
In the next InfiniTime release, the vibration will stop after a short time. Also if you don't use this feature, you can disable it in Gadgetbridge.

NCC1701M, 2022-03-23:
> In the next InfiniTime release, the vibration will stop after a short time. See #903.

This is definitely not the best solution. It has to stop when the call ends. I don't think that other devices vibrate endlessly or stop automatically. There has to be some API functionality.

pbryan, 2022-03-23:
There seems to be some debate on whether the Gadgetbridge connector has the required capability. API would be ideal, but at least this PR will keep our devices from vibrating endlessly until that can be resolved with a better solution.