InfiniTimeOrg/InfiniTime #368: Time format on Quick Actions screen does not honor Time Format setting

Batcastle, 2021-05-19:
I'm running Infinitime v1.0.0 and noticed a minor issue:

If you set the time format to 12-hour in the settings app, while the clock face does honor that setting, the clock in the top lefthand corner of the Quick Actions menu stays in 24-hour mode. I feel this clock should honor the same setting as the main clock face.

Raupinger, 2021-05-19:
Thanks for the bug report. I just fixed it in #370.

Riksu9000, 2021-10-02:
This hasn't been fixed yet so I believe this should still be open.

Batcastle, 2021-10-02:
I second that. The PR mentioned above is still open, and there have been no PRs mentioned in either the PR's chat or this one so this issue remains in the master/main branch. It should be reopened.

infinitytec, 2021-10-16:
Can confirm it is still an issue.

trman, 2022-02-21:
resolved by  #907 and #927  pr

Riksu9000, 2022-02-21:
Fixed by #821

closed 2022-02-21 by Riksu9000