InfiniTimeOrg/InfiniTime #373: Vibration motor doesn't stop.

MysteryLogger, 2021-05-19:
While I was at Work (From home...) my GF started saying that her watch (InfiniTime sealed model , I had given it to her has InfiniTime 1.0) started vibrating and wouldn't stop. 

What I have tried:
Resetting it (Long pressing the button)
Waiting for it to die and re-charge
Re-Flashing the OS.

All of these still cant fix the problem , the motor also seems to be getting weaker over time , maybe due to the vibration.

Any fix or should I buy a new one? Mine doesn't seem to be doing this (Mine is a devkit unit).

MysteryLogger, 2021-05-19:
Fixed it!! 

I flashed it using my Phone and it seemed to fix the issue (I also had to unpair it from the GadgetBridge app).

Is it a problem with the OS or the GadgetBridge App?

Avamander, 2021-05-20:
> All of these still cant fix the problem , the motor also seems to be getting weaker over time , maybe due to the vibration.

Very likely the battery just got drained really fast.

InfiniTime probably doesn't disable the vibration motor on reboot, maybe it should.

MysteriousLog6, 2021-05-21:
> What didn't work on her phone?

I think he means the software update.

JF002, 2021-05-27:
I'll close this issue. Feel free to reopen it if needed.

closed 2021-05-27 by JF002