InfiniTimeOrg/InfiniTime #374: Valid firmware resets to factory firmware on reboot

jarbus, 2021-05-20:
Same issue as #274, but my PineTime says my firmware is valid, and I've installed the firmware multiple times after each reboot, from 1.0.0 to 0.7.4

Avamander, 2021-05-20:
"Says its valid" in the settings after you've pressed "validate"?

What version of bootloader?

jarbus, 2021-05-20:
Yes, it says `You have already validated this version of the firmware`. I just installed the bootloader today, but I'm not sure how to check what version I installed; The file is called `reloader-mcuboot.zip`

jarbus, 2021-05-20:
Update: Using the reloader-mcuboot.zip file from https://github.com/JF002/InfiniTime/releases/download/0.14.1/reloader-mcuboot.zip, but it was having this issue with the factory bootloader as well.

JF002, 2021-05-20:
This should not happen... Let us check a few point:
 - Are you using a devkit (open) or a sealed device?
 - InfiniTime 0.7.4 does not exist, you probably meant 0.7.1, or are you referring to something else?
 - When upgrading, do you follow this procedure :
    * Upload the firmware using a companion app (OTA)
    * The watch reboots
    * The new version is installed, and you validate it by using the Validate app in the Settings menu
 - Did you install another firmware on this device (PineTime Lite, WaspOS) ?
 - Could you check the version of the bootloader : when the watch reboots, does it displays a green "PineTime" logo (old bootloader) or a white pinecone that is progressively drawn in green, with a version number on the bottom (new bootloader) ? See [here for more info](https://github.com/JF002/InfiniTime/blob/develop/doc/gettingStarted/gettingStarted-1.0.md#how-to-check-the-version-of-infinitime-and-the-bootloader).

jarbus, 2021-05-21:
Thanks for the assistance, @JF002. To answer your points:
- Sealed device
- Yep, 0.7.1, my bad
- I follow the procedure you describe, but the firmware resets after I reboot manually after the first reboot
- No other firmware installed
- Bootloader is latest version, displays the white pinecone that is filled in

JF002, 2021-05-21:
This is strange. The validation process is pretty simple : When you tap on the 'validate' button, a flag (a bit) is set in flash memory. When the bootloader runs, it checks this bit and revert if it's not set. 
Failing flash memory could be the cause of this issue, but the chances that only that specific bit is failing is quite low...

So uh... I've no other idea... Could you maybe shoot a video of the validation/reboot process?

jarbus, 2021-05-21:
Sure, let me know if I'm just doing something wrong

https://user-images.githubusercontent.com/42819002/119159813-9f0e4180-ba25-11eb-8b04-86d15cf77ca4.mp4

jarbus, 2021-05-21:
Also I wasn't able to get the time to update to the correct time, it always starts counting up from Jan 1 1970 at midnight

JF002, 2021-05-27:
> Also I wasn't able to get the time to update to the correct time, it always starts counting up from Jan 1 1970 at midnight

The time always resets to 0 (epoch, 1 january 1970) on reboot. You need to connect a compatible companion app over BLE to sync the time with your phone/computer.

I've just watched your video. When you reboot the watch, do you keep the button pressed  while it's displaying the white pinecone? I think I see it changing color to blue, which means that it's reverting to the previous version of the firmware.
Could you try to reset the watch and release the button as soon as you see this white pinecone? It should be drawn in green, meaning that it's booting normally, and it should not change color (to blue or red).

jarbus, 2021-05-29:
I've done as you described, same issue. However, when I set up the PineTime's firmware again so I could record footage of the same issue occurring, I got stuck in an infinite loop of the white pinecone filling up with green then resetting.

HomelyCactus, 2021-06-01:
I have the same problem. I got my sealed pinetime a week or so ago and installed https://github.com/JF002/InfiniTime/releases/download/1.1.0/pinetime-mcuboot-app-dfu-1.1.0.zip via Gadgetbridge. I was on v1.1 and looked the same as the start of the video until I rebooted the watch and I'm back on the factory version. 

My boot screen looks different though, I have the PINETIME in green letters, no the PineCone. My build is Jul 20 2020 Version 0.7.1

Avamander, 2021-06-01:
@HomelyCactus 
You have to confirm the firmware you've installed otherwise it gets rolled back next reboot.

Secondly, do update your bootloader to 1.0.0

HomelyCactus, 2021-06-01:
@Avamander I was pretty sure I did that but I could be mistaken.

Regardless, I cant do any updates, bootloader or firmware because anything I try to install I get "Element cannot be installed". On top of this my pinetime Doesn't stay connected to my phone for longer than a few seconds on first boot. I'll try a different phone I guess.

If I still have problems I'll open a new issue. I dont want to clutter up this one. Cheers

Avamander, 2021-06-01:
> Regardless, I cant do any updates, bootloader or firmware because anything I try to install I get "Element cannot be installed".

Use nRF Connect or wait for the next Gadgetbridge release.

jarbus, 2021-06-02:
> I've done as you described, same issue. However, when I set up the PineTime's firmware again so I could record footage of the same issue occurring, I got stuck in an infinite loop of the white pinecone filling up with green then resetting.

Letting the PineTime die fixed this issue, and my firmware does not reset anymore on reboot when releasing the power button immediately after seeing the pinecone.

closed 2021-10-02 by Avamander