InfiniTimeOrg/InfiniTime #376: Due alarm on connected smartphone leads to "null" notification

christianfl, 2021-05-20:
Hey! I just saw that an alarm set on my Android device (connected with Gadgetbridge 0.56.2 and Android standard clock app) causes a one time notification on the PineTime (InfiniTime 1.0) when its due - but that has no title and just the content "null" and gets dismissed like usual messages.

How cool would it be to show some interaction possibility (like with calls) that vibrates until its dismissed on the phone or watch? :-) Useful e.g. when the phone is in another room.

JF002, 2021-05-21:
Notifications are sent by the companion app (gadgetbridge in this case). InfiniTime just displays the content of the notification without trying to interpret the content. InfiniTime only ignore notifications that contains no data at all. In this case, Gadgetbridge probably sent a valid notification with "null" as message... I think this is something that should be handed in gadgetbridge.

Current implementation and protocole of notification does not provide any way to dismiss notifications after they were sent, but that's probably something that can be improved.

Avamander, 2021-05-21:
It would require a log of Gadgetbridge to find out what's creating these "null" notifications. Probably shouldn't be filtered out in the sense of forbidding "null" but instead it should be figured out what creates the "null".

christianfl, 2021-05-21:
Thanks for the quick answers. I will look a little bit into Gadgetbridge to see why there are null messages and how that could be improved.

JF002, 2021-05-27:
I'm closing this issue as it seems to be related to the companion app. Feel free to reopen it if needed.

closed 2021-05-27 by JF002