InfiniTimeOrg/InfiniTime #377: Disable Changing Clockface type with long press on Clockface

Batcastle, 2021-05-20:
This is mostly a quality of life improvement.

The idea behind this is that if the screen gets wet, or if someone isn't paying attention, this can very easily be triggered accidentally. A user not knowing about this function may think they have to go into Settings to change this, and if it happens a lot, it can lead to a lot of frustration. 

It would make a lot of sense to remove this ability, map it to something else, or add a setting to control it.

JF002, 2021-05-21:
The possibility to switch watchface with a long push was added before the setting app was integrated. I agree it makes a bit less sense now that we have a fully fledged Settings application.

I haven't noticed many occurrence of that issue on my unit, but I don't wear it under the shower ;)

What's the opinion of other users about this?

Raupinger, 2021-05-21:
I'm kind of split. I'm sometimes triggering it by accident, but I do still think it makes some sense. There have been a few issues related to water recently. What about a "lock screen" feature? Maybe bind that to a long press on the watch face and disable it by pressing the button.

Itai-Nelken, 2021-05-21:
My unit hasn't arrived yet so I don't know how annoying it is or isn't for me, but here is my opinion:
I think it does make sense to change watch faces with a long press, but maybe add a extra step so for example long pressing brings you to the settings with the clock face selection list. that way its harder to change the watch faces by accident but its still there.
maybe instead of being a shortcut to the list in the settings app, it can bring up a list of images of the watch faces that you can then select the one you want to set, though that will probably use to much memory and storage.

MaWalla, 2021-05-21:
That's probably just me, but I personally don't see a point in changing the watchface that quickly after making my initial decision. So I'm completely pro removing the long-press for changing it.

Batcastle, 2021-05-21:
I would like to point out that I can see the functionality of having a long press do SOMETHING. Someone on the Telegram chat suggested making it a configurable action for watch faces to use.

Also, this feature, implemented as it is now, might cause issues if/when support for more watch faces is added in the future. So it may be smart to remove it either way, depending on how the code works.

ptman, 2021-05-23:
I've been wondering why my watch face gets changed. This is super annoying.

schneidr, 2021-06-10:
IMO the long press should be removed. I often find my clockface changed after sleeping or when I had my hands behind my back 'Spock-style'. Additionally, it is becoming completely useless when more clock faces are added, then it's easier to change it through the settings.
I've been thinking about other uses for the shortcut, but I couldn't think of one. Maybe this can be used by a functional clockface or something like that.

JF002, 2021-06-10:
I think we can safely remove this feature now that we have a dedicated app in the setting menu to select the watchface.

PR #425 (2021-06-12): Remove tap to switch Watchfaces
resolves #377 
this also allows Clock to just keep a single screen pointer instead of a ScreenList
closed 2021-06-12 by JF002