InfiniTimeOrg/InfiniTime #378: Black screen and no bluetooth connectivity

serjor, 2021-05-20:
Hi!

I've just received my sealed PineTime, and during the first charge I've updated the firmware. Everything has gone smoothly, and the watch has rebooted without any issue.

The point is that after a while, during the first charge the watch has stopped responding, and now it does nothing. No Bluetooth, the screen is black and even if I try to hard reset the watch, nothing happens.

Duckduckgoing I've found that this happened in previous versions, but nothing about v1.0, and the "fix" was waiting until the battery dies.

Do I have to wait until the battery dies or there are other chances considering that the version is newer?

Thank you!

Avamander, 2021-05-20:
If it's a sealed device then the only way to truly reboot is to wait for the battery to run out. 

Make sure to follow the guide for best results: https://wiki.pine64.org/wiki/Upgrade_PineTime_to_InfiniTime_1.0.0

serjor, 2021-05-21:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEgAAABICAYAAABV7bNHAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAArSURBVHic7cEBDQAAAMKg909tDjegAAAAAAAAAAAAAAAAAAAAAAAAAAA+DFFIAAEctgHwAAAAAElFTkSuQmCC###24x24:true###][Imagen de rastreo][https://github.com/notifications/beacon/ABRQYTDNZD2SLEJQFE4JNB3TOWB3HA5CNFSM45H5NHAKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGJSX2TI.gif]

Ok, thank you.

So I'm afraid that I'll will start enjoying my pinetime in a week 🙄.
Not big deal, I was aware of what I was buying 👍

JF002, 2021-05-21:
Did you update the bootloader before the screen went black?

serjor, 2021-05-21:
I'm afraid I didn't. I overlooked the wiki and missed that part.

JF002, 2021-05-27:
So yes, if your device is still running on the older bootloader, that kind of issue can happen. Keep us up to date when the battery drains :)

serjor, 2021-05-27:
That was the plan, but when yesterday it came into life after pluging it to the charger, I tried the update the firmware to version 1.1

Unfortunately it didn't work. Right after the zip  file was downloaded into the watch it rebooted and only pinetime text appeared in the screen. As it was very late I left the watch in that state and this morning we are back to the black screen of nothingness.

This time battery wasn't fully loaded, so I hope that in a couple of days I can try again.

I'll keep you informed

JF002, 2021-12-30:
I'm closing this as there's no update for a long time. Feel free to re-open this issue if needed.

closed 2021-12-30 by JF002