InfiniTimeOrg/InfiniTime #380: Music player is sometimes shows the wrong playtime and stops working

MaWalla, 2021-05-20:
(this might also be a Gadgetbridge issue, I'm not sure to be fair :sweat_smile:)

When listening to music (using either the official Spotify app or others like my native player com.crdroid.music, which pretty much is the same as the default LineageOS one), InfiniTime's music app sometimes displays a wrong playtime (e.g. 19:12/5:04 or 87:05/3:15 despite the song just started playing). Sometimes it seems to go on, sometimes its just stuck at the value.

When the time is wrong, sometimes the buttons (last, play/pause, next) just stop working altogether, or just skip like 5 songs at once.

I haven't yet pinpointed what exactly needs to be done to work around the issue, but so far I've been fairly successful with it by just briefly pausing and resume playing the song on the phone.

I'm using Infinitime 1.0.0 with Gadgetbridge 0.56.2 from F-Droid on my OnePlus 6T (running crDroid, which is based on LineageOS)

Avamander, 2021-05-21:
Wrong playtime would indicate that the music player doesn't broadcast new correct position or the packet gets lost. This malfunctioning music player can probably detected somehow, but it's probably easier to just use a better player.

Lagging buttons indicate a lagging connection, given the small amount of traffic between the phone and watch in general, I would point fingers at the Bluetooth stack crapping out on the phone (instead of queue-bloat). Diagnosing this would require an Android log (`adb logcat`) with matching timestamps of your own actions on the watch.

joeybronzoni, 2021-07-13:
I've had similar issues with recently installing version 1.2 where the music player stopped working with LineageOS and also using Gadgetbridge for the connection.  It was working in version 1.0 for me -for the record.

Riksu9000, 2022-03-28:
Sounds like the same issue as #127. What do you think @MaWalla?

Avamander, 2022-04-11:
Let's reopen if this is still an issue.