InfiniTimeOrg/InfiniTime #381: Is fabtolab.com trustable

MysteriousLog6, 2021-05-21:
I am interested in trying to get a PineTime DevKit but sadly the official website doesn't ship to India another website says that they do it is fabtolab.com .

Are they trustable?

Avamander, 2021-05-21:
Probably not a question for InfiniTime, rather the forums or chat.

JF002, 2021-05-21:
I think you're not alone, I've heard from other people from India who have a hard time to get Pine64 devices shipped to their country, unfortunately.
As @Avamander, your question is not directly related to InfiniTime, and I would suggest to ask your question on Pine64 forum and/or chat rooms.

MysteriousLog6, 2021-05-21:
> I think you're not alone, I've heard from other people from India who have a hard time to get Pine64 devices shipped to their country, unfortunately.
> As @Avamander, your question is not directly related to InfiniTime, and I would suggest to ask your question on Pine64 forum and/or chat rooms.

I have asked in the forum , never heard of the chat rooms.

closed 2021-05-27 by JF002