InfiniTimeOrg/InfiniTime #383: Feature Request: Runners App

PetersOtto, 2021-05-21:
## Idea "Runners App"
If you are running a new track or if you are running in a competition, it is very helpful to know how many kilometres you had run.
My idea is to have a runners-app that shows the running time (Stopwatch) and the kilometres you had run.
Professional Sportwatches have GPS but perhaps it is possible to use the step counter for calculating the distance.

## My Track
At first I just had some test runs on my track at home. I picked some positions on that track and measure the distance with the sportwatch from my wife (Garmin Forerunner 735XT). After that I ran the track with my PineTime (InfiniTime 1.0.0) and pick up the steps.

### Measurement (GPS)
Date       |  Pos1 / km | Pos2 / km   | Pos3 / km | Pos4 / km | Pos5  / km | Pos6 / km
-----      | ----- | ------ | --   | -------- | ------------ | ----
16.12.2020 | 0     | 1,36   | 2,23 |  3,53    |  4,53        | 5,81
18.12.2020 | 0     | 1,31   | 2,17 |    –     |  4,47        | 5,75
20.12.2020 | 0     | 1,32   | 2,20 |  3,51    |  4,49        | 5,76
21.12.2020 | 0     | 1,40   | 2,29 |  3,60    |  4,59        | 5,86
24.12.2020 | 0     | –      | 2,26 |  3,52    |  4,55        | 5,75

–         |  Pos1 / km | Pos2 / km   | Pos3 / km | Pos4 / km | Pos5  / km | Pos6 / km
-----      | -----           | ------            | --               | --------        | ------------   | ----
average | 0,00	       |1,35              |	2,23  	|3,54	   | 4,53	      | 5,79

### Measurement (running with PineTime)

Date       |  Pos1 / steps | Pos2 / steps   | Pos3 / steps | Pos4 / steps | Pos5  / steps | Pos6 / steps
-----      | ----- | ------ | --   | -------- | ------------ | ----
22.04.21 | 0 | 1300	|	2140	|	3410	|	4398 |	5736
25.04.21 | 0 |1346	|	2254	|	3576	|	4610 |	5978
10.05.21 | 0 |1312	|	2182	|	3464	|	4478 |	5854
12.05.21 | 0 |1312	|	2198	|	3514	|	– |	5808|
14.05.21 | 0 |1300	|	2134	|	3380	|	– |	5592|

### Measurement (walking with PineTime)

Date       |  Pos1 / steps | Pos2 / steps   | Pos3 / steps | Pos4 / steps | Pos5  / steps | Pos6 / steps
-----      | ----- | ------ | --   | -------- | ------------ | ----
14.05.21 | 0 | 1780	|	2896	|	4570	|	5860 |	7518

## Summary

In my opinion it is possible to have a runers-app. However the measurement is very individual and depends from body structure of the user and how fast she/he runs. So the app need a calibrating mode.

Perhaps like this: The user starts a calibrating run, runs a distance (400m - light athletics place at school) with normal speed, stops the time and input the distance. After that the app knows the normal running speed and step length of the user. If the user is faster, the app calculate with a bigger step length. If the user is slower, the app calculate with a shorter step length.

## Next

If there is an Interest to code this app, I could do some test runs on a 400m light athletics place.

Perhaps just walking, running with normal speed and run so fast I can.
With taking the time and the steps, it is maybe possible to develop a app?

With code I can't help, but I will run for you :)

What do yo think?

MysteriousLog6, 2021-05-22:
> but I will run for you :) 
This made me laugh :)

Talking about the app , yes it is very much possible to do this theoretically but the device seems to have a small amount of ram and internal storage (Will be using an extra 4mb storage that is just sitting) we will have to make an app and then test the amount of ram it takes.

We could also merge it with the steps app , we could also add an option for custom stride length (Got the idea from Garmin :) in which you walk a set distance (You can customize the said distance) and then count the steps it took you to do that then input it and the PineTime now knows a more accurate measurement to use.

This is basically the calibrating mode you had mentioned.
But with the extra step that you can adjust the distance and step count is your own calculation (Might be a problem)

PetersOtto, 2021-05-23:
Dear MysteriousLog6 
Thank you for your feedback!

> Talking about the app , yes it is very much possible to do this theoretically but the device seems to have a small amount of ram and internal storage (Will be using an extra 4mb storage that is just sitting) we will have to make an app and then test the amount of ram it takes.

I am sorry, I think I can't help here. :(

> We could also merge it with the steps app , we could also add an option for custom stride length (Got the idea from Garmin :) in which you walk a set distance (You can customize the said distance) [...]

In my opinion the solution you got from Garmin is good and in most cases enough! My Idea was to increase the accuracy by using the Stopwatch. 
If you just walking/hiking (0,770 m/step) the stride length is nearly 24% shorter than on a normal run (1,008 m/step). By using the time it could be possible to detect if the user is walking or running.

MysteriousLog6, 2021-05-24:
Well we can detect the running or walking by checking the speed of the accelerometer..

The custom stride length is useful as the person can adjust it to be any length all we have to do is some quick math.

PetersOtto, 2021-05-24:
The accelerometer, of course! That is a good idea.

However it is important to know the time you did run.
Perhaps it is possible to combine the stopwatch app and the runners app and show both on one screen?

bearclaws8, 2021-08-13:
One question on scope for such a feature, is the desire to have only the PineTime on the user?  I was thinking that this could be a great feature for running/biking, all of which could use the same display layout with different mode selections.

One possibility is leveraging data from an existing phone app and just displaying it on the PineTime.  That may help with data storage limitations on the PineTime for large tracks.  I have used an open source tracker, [OpenTracks](https://f-droid.org/en/packages/de.dennisguse.opentracks/) that is available on Fdroid on my phone.  It captures much of the information you are looking for and can show it via notifications on the phone.  I'm not sure if it has a way to pull the data from it but if the PineTime could snoop the notification it may be able to display the data (and maybe stop the notification from showing up in the notification area of the watch).  This may be limited in the number of users that can benefit though.  

However, if you only want the PineTime to be worn this idea may not work for you.  Hopefully my thesis above isn't out of scope for your feature request.

Unfortunately I too can't help with code that much but ideas I have in abundance.  And a willingness to try and help test, performance while running may be limited...

iameru, 2023-01-23:
I also would love to have the connection to Opentrack and display some current data from opentrack on the watch!
I have no experience in C or C++ though but still would love for an experienced person to give some insight in how much work this would be. If it is "just some linking up and data parsing" I might be willing to give it a try. In the end this would need an extra screen / app on the watch.