InfiniTimeOrg/InfiniTime #389: Add compiler flags : -fno-rtti, -fno-exceptions, -fno-non-call-exceptions,...

JF002, 2021-05-22:
I really thought these flags were already enabled but... it looks like they are not!

We are in an embedded system, we don't need some high level functionalities like exceptions and RTTI. Disabling these functionalities at build time should reduce the size of the final binary.

I think we should be able to add `-fno-exceptions -fno-non-call-exceptions -fno-rtti` to `COMMON_FLAGS` in src/CMakeLists.txt.
It might be interesting to experiment with other flags mentioned here : https://bitbashing.io/embedded-cpp.html

Avamander, 2021-05-22:
:S RTTI might exactly be the thing that I'm currently building upon. I'll investigate. The rest, possibly. 

Though lack of them does make debugging harder in any case I think.

Avamander, 2021-05-23:
This issue maybe should be about compiler flags in general?

I'd pull a few other topics from other issues to here:

`Os` has been mentioned multiple times, but I have to warn that it can turn an usable UI into an unusable one. Performance regressions are a real risk and it can't be blanket-applied. I've shot myself in the foot with this with a few other projects. So if at all possible, let's please leave stripping out code and space-optimizing compilation as the last straws.

`LTO` has also been mentioned in a few other issues. Someone even tried it, but didn't manage to achieve a fully working configuration. It really isn't trivial but the result can be great.

JF002, 2021-05-24:
> :S RTTI might exactly be the thing that I'm currently building upon

dynamic_cast?

> This issue maybe should be about compiler flags in general?

Yes, I think this is a good opportunity to investigate all compiler flags that would be useful for the project :)
And I agree with you, we won't apply flags and just be happy by the code size reduction! There's certainly good reasons to not enable these flags by default and we must be aware of potential side-effects.

`Os` (already mentionned [here](https://github.com/JF002/InfiniTime/issues/335)) and `LTO` are indeed good candidates!

I've done some quick tests: Add those flags in `src/CMakeLists.txt` at line 747 (`set(COMMON_FLAGS ...`):

 - original size : 426.512 B
 - `-fno-exceptions -fno-non-call-exceptions`: 414,264 B (-12,248 B)
 - `-fno-rtti`: 422,312 B (-4,200 B)
 - `-fno-exceptions -fno-non-call-exceptions -fno-rtti`: 409,320 B (-17,192 B)
 - `Os` for **the entire code** : 358,148B (-68,364 B)
   * only on nrf-sdk : 421,896 B (-4,616 B)
   * only on nimble : 399,560 B (-26,952 B)
   * only one lvgl : 396,024 B (-30,488 B)
   * only on pinetime-app : 403,148 B (-23,364 B)
 - All of the above combined : 340,236 B (-86,276 B) -> ~20% of binary size reduction!

Note that I haven't run any of these builds on an actual hardware. Anyway, these numbers give good indications on potential gains !

JF002, 2021-05-25:
@Avamander I'm curious : why do you need RTTI? And also, how removing exceptions and RTTI would make debugging harder? Except for 1 dynamic_cast, we are not using RTTI for now, and we are using no exceptions at all.

Avamander, 2021-05-25:
> @Avamander I'm curious : why do you need RTTI?  

Need? Maybe not, I were using dynamic_cast :D

closed 2021-06-01 by JF002