InfiniTimeOrg/InfiniTime #390: Improve setting the timer value

km6thj, 2021-05-23:
1. Three cheers to the timer! A great new feature!
2. Setting the time in the timer is somewhat awkward on my watch at least. I have to wait at least 1 second between presses, so it takes about 30 seconds to set the timer to 30 seconds. One proposal would be to keep incrementing/decrementing the value when one of the + or - buttons is held. Another would be to set the time faster by swiping up and down.

Raupinger, 2021-05-23:
one idea i had were +10/-10 buttons, but i felt that would have become to crammed

Raupinger, 2021-05-23:
I guess that could be mapped to swipes

Soundtoxin, 2021-05-23:
Not sure if this should be its own issue, but I was thinking having saved timer presets would be good as well. The Pebble had them. So, if you commonly use a 5 minute timer, or you use an 18 minute timer for your tea to cool every morning, you can just select it from a list and start it instead of manually inputting a specific time for something you use a lot.

Avamander, 2021-05-23:
Pebble Timer watchapp UI: https://www.youtube.com/watch?v=gpImEfTlQ44

Raupinger, 2021-05-23:
So how about this: Swipe up and down to add or subtract 10 seconds. Swipe right to see a list of presets and one "+" item to add the current value as one. Still needs a way to delete presets since I'm gonna have to put a limit on how many you can have.

Soundtoxin, 2021-05-23:
> So how about this: Swipe up and down to add or subtract 10 seconds. Swipe right to see a list of presets and one "+" item to add the current value as one. Still needs a way to delete presets since I'm gonna have to put a limit on how many you can have.

That sounds pretty good to me. Is there a known reason you can't press the buttons faster to increase the time, though? Can the responsiveness be improved in the future?

km6thj, 2021-05-24:
Is it possible to have something like press-and-hold in infinitime?

km6thj, 2021-05-24:
Related note, is it also planned to add hours to the timer, or even days?

Raupinger, 2021-05-24:
@km6thj 
The answers to your questions are: "probably no" and "most definitely no". Let me elaborate:
The info apps get on touch events is rater limited. They don't know how long exactly a button has been pressed, just wether it was long or short. There might be a way to get that info out of the current setup but I wouldn't know it.
The underlying hardware counter the timer uses rolls over about every 4 hours. The way the Timer is currently built multiple hours on a timer just aren't possible. You could internally split a timer up into multiple with each starting the next, but that sounds kind of stupid. Also, the text size would have drop down from 72px to 42px making the timer harder to use in general. I think the better solution in that scenario is to use the Alarm app I'm planing to build soon™.

km6thj, 2021-05-27:
Thanks @Raupinger! For me personally, the digits could be smaller -- but that is subjective of course. Having a timer of up to 4 hours would definitely a win over one that can have max. 1 hour, again for me personally :)  Like when you put a long cooking meal into the oven for Christmas, for example,  it's really useful to have 2 or 3 hours timers. I can't speak for everyone for sure, so not sure how many use cases there are really for this. I always found it useful on my iphone and other devices.

heinrich-ulbricht, 2021-08-09:
Just wanted to note that my old Vector Watch remembered just the last timer setting. That was all and it was good enough most of the time. So. Remembering _anything_ would be a plus for me :)

Riksu9000, 2021-08-28:
This situation got better with #492 getting merged, which makes touch much more responsive. Now it is also possible to make holding the button down increment the value. I might add some more improvements to #554.

closed 2021-10-02 by Avamander