InfiniTimeOrg/InfiniTime #391: Timer ending is not obvious enough

Soundtoxin, 2021-05-23:
From what I understand, when a timer runs out, it vibrates once and then resets to 00:00.

- It should vibrate repeatedly until dismissed so you don't ignore it and have your water boil over or similar (you may be timing something very urgent)
- There should be some sort of message that your timer expired so if you look at it a minute late you remember what happened

I also think it should not clear the timer value, but maybe that would work best after implementing my idea (https://github.com/JF002/InfiniTime/issues/390#issuecomment-846558561) to save timer presets. It would then show something like 05:00 after your 5 minute timer ended, and you would press the back button to get to a screen that lists all your timers, similar to the Pebble timer app.

Raupinger, 2021-05-23:
I wanted to have continuous vibrations for the initial release but they depend on #342 getting merged. As soon as that's in this should be trivial to change

km6thj, 2021-11-03:
#342 went in it seems -- any chance this will also be resolved?

hatmajster, 2021-11-03:
What about: https://github.com/InfiniTimeOrg/InfiniTime/pull/557 ?

Lacplesis, 2021-11-11:
I also want to throw in my support for this feature and offer a suggestion. Maybe in the settings, you could have customizeable vibration patterns for the apps, such as **Short Vibration, Long Vibration, Tripple Vibration, or Continuous Vibration**. I find it very easy to miss the single short vibration if I'm moving around at all and this would solve that problem.

The-Linux-User, 2022-04-23:
Just wanted to showy support for this feature, it's so easy to miss the timer going off when it only vibrates once

Fidgetcetera, 2022-06-29:
Echoing the support for this, I was just looking to make an issue when I found this one. I can't use the timer on my Pinetime at all because it's really not reliable whether I'll feel my wrist buzz (or if I'll think it's a notification and ignore it because I'm cooking something and it Can Wait)

mashuptwice, 2022-07-28:
@Riksu9000 @JF002 Is there any work planned on this issue? It is now open for well over a year and even tho I've found a few PRs which claimed they would fix that issue, all of them seem like they were abandoned or lost in discussion. #557 is still umerged 6 months after the last commit, it seems because of a similar race condition as the one between notifications and calls. 

It looks like the single vibration breaks functionality for quite a few people, would it be possible to merge #557 with the currently existing problems and afterwards work on a better implementation?

Riksu9000, 2022-07-28:
We tend to work on things that interest us and the timer has been working fine for me for short durations. This should be pretty easy to implement since you can reference the Alarm app heavily, so anyone's free to contribute this feature.

mashuptwice, 2022-08-11:
On my [InfiniTime Fork](https://github.com/mashuptwice/InfiniTime) the default timer vibration is 1000ms long. That fixes the problem for me until someone works on a more perfect solution.

Avamander, 2022-10-04:
I think Timer/Alarm and vibrations have been changed a few times now, if this is still an issue let's reopen.

Soundtoxin, 2022-10-04:
@Avamander Zero of the mentioned issues in the original post have been fixed as far as I can tell. It still vibrates just once gently and then shows no sign that a timer has expired. Please re-open.

Riksu9000, 2023-01-15:
I use the timer daily and I like the fact that I don't need to interact with the watch when it alerts. If I can't address it immediately, I can simply start another timer. Therefore I think it might actually be best to simply add a short vibration pattern, but not make it ring continuously.

Soundtoxin, 2023-01-15:
That would go against standard timer behavior seen in Android as well as on the Pebble watches. You should be able to set a 30 minute timer and have it wake you from a nap, and a timer should be able to properly get your attention even if you're busy so that you don't burn a roast in the oven or have your soup boil over or similar. For low-urgency reminders the current behavior might be fine, but I think it's better to account for more possible scenarios. I would say maybe it could be configurable, but I'm not sure how that would work with so few buttons on the watch and not too deep of menus.

The main difference between a timer and an alarm the way I see it is using relative time instead of absolute time. Sometimes "I'll check it again in a half hour" is easier than checking what the current time is and figuring out exactly when to create an alarm for. Timers are general purpose reminders.
Following the timer ~= alarm example, it would be quite bad if your alarm to wake up for work just vibrated gently once and did nothing else. Timers can be just as urgent as alarms in my experience and aren't to be taken lightly.

Riksu9000, 2023-01-16:
Android isn't exactly comparable, because a phone isn't strapped to the wrist, and as far as I can tell, pebble didn't come with a timer. Not that it matters, because we should not simply copy others.

My suggestion is to tune the vibration such that it won't be missed while awake, and making it continuous wouldn't make a difference at that point.

Using the timer to wake up from a nap is a fair point, but if we want to account for more scenarios, why not use the alarm for wakeup, as intended, and have the timer not require interaction, so it won't be a hindrance while you're already busy doing something, such as when cooking.

mashuptwice, 2023-01-16:
@Riksu9000 

I personally also don't like the idea of having a continuous ringing timer, but:

I have seen the statement to not "copy" from android or other OS a few times now on this repo. Using this as an argument when apparently only two options exist, continuous or self-stopping timer vibration, seem like an easy way to hinder any progress.

I can't see how it is a valid argument to not do one thing just because android is doing it this way. It is a fact that in the tech world nearly everything has been done by one or the other and the beauty of OSS software is being able to reuse existing concepts and even code across multiple projects.

Imagine a car manufacturer only producing cars with 5 wheels, because 2, 3 and 4 wheels were already done by others.

Riksu9000, 2023-01-16:
@mashuptwice I think you've got it backwards. I never said we should not do the same thing as others. Instead, I don't see how saying we should do it this way because someone else did it this way is a valid argument. My argument is that a phone is not the same as a watch.

JustScott, 2023-06-13:
I had a timer set for 2 minutes when I came to make an issue for this... and I missed the timer while typing ( didn't even notice the vibration ). I think the timer should have the same non-stop vibration as the alarm does. This is one of my only issues with the Pinetime, other than that I absolutely love this device.

jarinox, 2023-07-05:
Would it be an option to add a setting to choose between those two behaviors?

Personally I would rather prefer a timer that rings until I stop it as I currently most of the time don't even notice the timer to have finished. Just making the timer buzz a little longer wouldn't resolve the problem of not noticing it for me as sometimes I am in a really deep focus. It would be essentially for me to have to stop it in order to rely on the timer. However I see how to other people might prefer the current behavior and not needing to stop the timer.

I think as we see the benefits of both options I think one should be able to choose which one to use!

borkymcgee, 2023-07-15:
> That would go against standard timer behavior seen in Android as well as on the Pebble watches.

Forget Android and pebble, every device I've ever owned with an alarm or timer feature has either:
- Rang for at least a few seconds
- Rang continuously until the alarm is dismissed

As is, the lack of in insistent alarm or timer is one of the few things keeping me from recommending this watch to others, as it lacks this simple feature that every dumb digital watch i have ever owned has had.

spaetz, 2024-01-05:
> We tend to work on things that interest us and the timer has been working fine for me for short durations. This should be pretty easy to implement since you can reference the Alarm app heavily, so anyone's free to contribute this feature.

Given this comment, I find it kind of amusing that PRs contributing the feature have been closed without even looking at the code. I'd think that PRs making the alert last for 10 seconds are a pretty tolerable compromise.

A long-termish solution could be to make alarms settible in relative times. (set alarm in 25 min)

DavisNT, 2024-06-30:
Just created #2085 - yet another PR for fixing this (a super-simple one which just vibrates until dismissed by button or sliding down on the screen). 
P.S. This PR was inspired by #1499.