InfiniTimeOrg/InfiniTime #393: Improvements to "Find my Watch"

Avamander, 2021-05-23:
The current "Find my Watch" functionality utilizes the call notification functionality. I think it wouldn't be a bad idea to separate this functionality into a separate watchapp **alongside** "Find my Phone" feature (https://github.com/JF002/InfiniTime/issues/343).

The current implementation has the downsides that:
* The UI doesn't necessarily represent the use
* Timeouts (e.g. screen turning off) are not optimal for the functionality
* Vibration pattern or length is too short for locating the watch

Raupinger, 2021-05-23:
The lather two shouldn't be an issue once  #342 gets merged, but I agree that there should be proper UI for this

JF002, 2021-05-27:
I think gadgetbridge send a "real" phone call (to the point of view of InfiniTime) when the feature Find my Watch is activated, and that's why InfiniTime displays a call notification ;)

But yes, I agree that it's not the most relevant UI for the functionality. I can't remember if the Alert Notification Service / Find My Device profile define a specific notification type for this functionality but if it's the case, that would be trivial to implement it in InfiniTime and companion apps.

MysteriousLog6, 2021-06-09:
> I think gadgetbridge send a "real" phone call (to the point of view of InfiniTime) when the feature Find my Watch is activated, and that's why InfiniTime displays a call notification ;)
>
> But yes, I [...]

This is true , the mi band 5 (Supports gadgetbridge) gets a call from the gadgetbridge app this is probably true for the PineTime too!

Thus GadgetBridge has to change it or we have to do some blackmagic.