InfiniTimeOrg/InfiniTime #394: Feature Request: Display an hours column in stopwatch

Soundtoxin, 2021-05-23:
Currently if the stopwatch goes past 60 minutes, it will show you're at 70 minutes instead of 1 hour 10 minutes. I think timing things in the hours is fairly common (speedrunning) and it would be good to have this extra column. I believe the Pebble displayed a # of hours.

Avamander, 2021-05-23:
The Pebbles display years if you leave the stopwatch running that long, heh.

grayrattus, 2021-05-24:
I'm not sure if adding hours is a good idea. Currently, the vibration is soo light that if you set a stopwatch for like 20 min you can easily forget that you set it and you won't even feel that it was vibrating. Also adding more numbers to the screen means it will be harder to set it up. Maybe it's only with my PineTime but not every touch responds to the time change in the stopwatch app :|

If it has some custom vibrating style for example vibrating JaJa Ding Dong then it would make sense. Or stronger vibration in general. I can't imagine after 1h of training to feel current vibration on the wrist :D

Riksu9000, 2023-02-20:
Instead of simply shrinking the font to add hours, the time display should adapt to the required number of digits.

JF002, 2023-04-16:
This was added in #1692.

closed 2023-04-16 by JF002