InfiniTimeOrg/InfiniTime #395: Feature Request: Raise Wrist to wake, but Double Tap to unlock

Soundtoxin, 2021-05-23:
It is useful to raise your wrist to see the time, but currently the setting is too sensitive and I keep doing random actions on my watch by accident throughout the day. If I could check the time at a glance, but still have to tap to unlock (allow actions), this would feel like a good combo.

I could see a couple ways of implementing this.
1) Raise Wrist displays the currently-open app whether it's the clock or not
2) Raise Wrist displays the time specifically, like a screensaver or Android lockscreen

After thinking about it a bit, I'm leaning toward the second option.

MaWalla, 2021-05-23:
I'm definitely agreeing on this (specifically option 2). My PineTime does lots of random things throughout the day, and when raising the wrist, I'm sometimes in some menu all of sudden.

This could be a great combination with #377 since long-press on the watchface could be used to unlock/manually lock the screen for example (which could further be indicated by a lock icon in the top left corner since we have plenty of space there, at least for the digital watchface), though I could also see the use of the physical button here in some way.

Furthermore I could also see settings for seperate brightness/screen-on time for the lockscreen, since there is a fair amount of "accidental" wakeups, especially with the raise writst to wakeup option. Being able to lower brightness and duration here would help with battery life a lot and is often sufficient for a quick glance.

Riksu9000, 2023-02-20:
Let's track this issue in #618.

closed 2023-02-20 by Riksu9000