InfiniTimeOrg/InfiniTime #397: best way to gather more data to improve heart rate accuracy

limoncelli, 2021-05-23:
As the heart rate sensor is still a bit unreliable, it may be improved by using a community approach gathering a lot of data and building a prediction model from it. A discussion on how to get hr data from pinetime has been initiated here https://github.com/alexr4535/siglo/issues/33 @ siglo and daniel has been very swift in responding here https://github.com/daniel-thompson/wasp-os/issues/204. So even though downloading data as raw as possible from the pinetime is possible there is still the need to gather the true actual hr eg measuring ones pulse for 60secs. This is currently quite difficult, max screen timeout after 60secs, no cue when measurement of hr starts @ infinitime. So the easiest would be the latter in addition to an increased timeout, though a  better solution would be a temporary heart rate collection app, where you are asked to store your heart rate measured by hand after 60secs infinitime collecting data.

IanWardell, 2021-05-26:
+1 , also I'm interested in getting the data from the pinetime to a mobile device so it can be used in fitness apps. I have no idea where the community stands on this currently in terms of implementation.

MFAshby, 2021-05-29:
Maybe daft suggestion: 2 watches? Collate data from a few other brand devices (Fitbit?) worn simultaneously with the pinetime. Would need some volunteers with other wearables but no doubt a few such people exist.

limoncelli, 2021-05-30:
I think there are two separate parts to it

a) getting the sensor data and
b) having some way to sync with any type of external measurement

Yes, for b), certainly another device would be helpful, like we have seen it for validating the step counter. But you'd still need the possibility to sync the measurements somehow.  So in my view, in the end you want something like

Date (60secs)       ; infinitime heartrate; real heartrate; raw sensor data
2021-05-29 18:03; 57                               ; 64                  ; 34, 37, 69, 60, 36, 38, 71, 70, 34, 35, .... , 38, 35

The real heartrate would be added manually after each 60 secs o f measurement, eg by looking at your fitbit or by taking your pulse. But you will need some indication by infinitime when the measurement starts.

MysteriousLog6, 2021-06-07:
On the Mi Band there is an option for tracking heart rate very minute so it is possible to do the test with someone who has both watches I have a mi band but not an PineTime :( so I cant help with the PineTime part but can help with the Mi Band Data stealing(Borrowing).

wilsonjwco, 2021-07-20:
Even without the ground truth, having raw sensor data along with InfiniTime's estimate of HR can help. It should be straightforward to see the HR in an FFT or spectrogram ([see this paper](https://www.researchgate.net/figure/Spectrogram-calculated-from-raw-PPG-measurements-compared-with-ECG-reference-value-black_fig1_303276918)) My vote would be to prioritize logging and downloading raw sensor data first, then figure out practices for syncing with ground truth later.

MysteriousLog6, 2021-07-20:
I have one method that could be done:

1) Use two watches and log info

Most watches have a file with heart rate data logged we could use these files and extract data from the file of another watch , InfiniTime could have a firmware with logging increased to maximum and then compare the two.
Comparing can be done using the time info given in the logs which has to be synced, Mi Bands do have the time info inside the log InfiniTime could also do the same with the log (Which includes raw sensor data) the Mi Band however is also not accurate (We could use one of the scientific YouTube reviews which has compared it with a proven heart rate device) to help the data. 

We can then use this data and then compare it with the "polished" data (The one that is compared with the reviews) and make a chart , using this chart we could keep tuning until it becomes very similar.

The logging on InfiniTime could be done by making a log file which is loaded to ram as a buffer  , once ram is filled it could be sent to the companion phone through Bluetooth.

There are open source tools that allow us to view the log data of the MI Band (Stored in Mi Fit) I could make a tool that makes this much more seamless.

Please do check if these ideas are plausible or far fetched.

stevensu1838, 2022-10-26:
Hi all, 

I've got two questions about heart rate accuracy. Can you experts share your insights? Many thanks. 

First, I've noticed the heart rate reading from the **Pine Time watch sometimes goes up to 160 for no reason**. Do you have any solutions? 
Secondly, may I please ask if you think the accuracy of the heart rate measurement is reliable? I am choosing between Pine Time watch and Garmin watch for heart rate  reading for research purpose. The advantage of Pine Time is open source and free for data streaming. However, It is charged for live data streaming with Garmin watches.

nxdefiant, 2023-01-11:
I assume this problem still exists with pine time? Maybe you could team up with some folks at Bangle.js  [as they have the same problem](https://forum.espruino.com/conversations/372681/?offset=50#16454552).

Avamander, 2023-01-11:
@nxdefiant We have a PR in the works that improves the current situation.

agx, 2023-04-12:
@Avamander is that PR linked somewhere?

FintasticMan, 2023-04-12:
#1486

closed 2023-04-30 by Avamander

marcinkosztolowicz, 2024-10-15:
I created a fork where, among other things, I implemented a Bluetooth service that exposes raw PPG data. Additionally, this fork includes the ability to configure the HRS3300 sensor, which is the hardware responsible for collecting the data. There are also Python notebooks—one allows data collection, while the other demonstrates basic data cleaning and visualization on a chart. The analysis is currently at the proof-of-concept (POC) stage, but it might already be of interest to someone.

https://github.com/marcinkosztolowicz/InfiniTime/blob/heart-rate-debug-tools/FORK_DOC.md

https://github.com/InfiniTimeOrg/InfiniTime/discussions/2113