InfiniTimeOrg/InfiniTime #398: Screen Timeout handling issue

Batcastle, 2021-05-23:
Screen time out is reset only when initial contact is made with the touch screen. In situations like the Pong game, this means the screen times out in the middle of the game because the finger may not have been lifted from the screen. I think that either:

 A) screen time out should be outright disabled in some situations.

or

B) the screen should not time out while contact is being made with the screen.

I can see the benefit of either approach, but think the first one would be better from a battery life stand point.

Raupinger, 2021-05-23:
there are already instances where sleeping gets disabled, (the flashlight for example) Its just a matter of identifying where the screen should stay on and adding that line of code there.

Batcastle, 2021-05-24:
@Raupinger Awesome. I'm glad it's already been handled then. I noticed this issue while playing with the Pong game so could this be fixed in that app then?

MysteriousLog6, 2021-06-01:
> @Raupinger Awesome. I'm glad it's already been handled then. I noticed this issue while playing with the Pong game so could this be fixed in that app then?

I guess so , will be checking about this and maybe making a PR.

Riksu9000, 2021-09-12:
The screen won't time out while touching the screen anymore after #492, which is in InfiniTime 1.4.0, so this can be closed.

closed 2021-10-31 by Riksu9000