InfiniTimeOrg/InfiniTime #418: Setting ReturnApp TouchEvents to Tap causes weird behaviour

Riksu9000, 2021-06-07:
If a SwipeDown event was changed to a Tap event in DisplayApp.cpp, the tap would "fall through" to the previous app. I found this problem when I tried to make the FlashLight app close on tap.

In this video
`ReturnApp(Apps::Settings, FullRefreshDirections::Down, TouchEvents::SwipeDown);`
was changed to
`ReturnApp(Apps::Settings, FullRefreshDirections::Down, TouchEvents::Tap);`
on line 314

https://user-images.githubusercontent.com/37774658/121019893-82edfc80-c7a8-11eb-892a-5bb6d97bf39a.mp4

JF002, 2021-06-10:
It's probably because the tap event was registered by the touch driver (https://github.com/JF002/InfiniTime/blob/develop/src/displayapp/DisplayApp.cpp#L377) and then processed by DisplayApp (https://github.com/JF002/InfiniTime/blob/develop/src/displayapp/DisplayApp.cpp#L185).

You can try to implement `OnTouchEvent(TouchEvents event)` in the Flashlight app and manually exit the app with a tap event is detected.

Riksu9000, 2021-06-11:
I think I've found the issue. It is like you said. `lvgl.SetNewTapEvent` gets run even if the app or DisplayApp processes it, creating an lvgl tap which activates a button later. The solution would be to only run `lvgl.SetNewTapEvent` if the tap won't get processed elsewhere.

PR #426 (2021-07-24): Fix TouchEvent::Tap
Don't run `lvg.SetNewTapEvent` if the tap was processed elsewhere.
This fixes #418
Now `OnTouchEvent(TouchEvents event)` must return false if TouchEvent::Tap wasn't processed. QuickSettings and Flashlight misused this and they have been fixed.
closed 2021-07-24 by JF002