InfiniTimeOrg/InfiniTime #421: Adding the ability to change the watch face without reflashing InfiniTime

CoderThomasB, 2021-06-08:
I was wondering if it was possible to send new watch faces over Bluetooth, and use a checksum to make sure the watch face has been sent correctly. Then if the checksum fails to use a fallback watch face.

Would this be possible to implement and would it cause issues if it was?

By the way, I don’t own a PineTime and therefore can’t write or test any code. Would anyone else be interested in this potential feature?

MysteriousLog6, 2021-06-09:
> I was wondering if it was possible to send new watch faces over Bluetooth, and use a checksum to make sure the watch face has been sent correctly. Then if the checksum fails to use a fallback watch [...]

Could be possible in the future using the LittleFS Filesystem (Still trying the tests)

CoderThomasB, 2021-06-09:
I think having a file system woude make this much easier to implement!

Raupinger, 2021-06-10:
Its actually much harder than that. Let me explain: 
This would mean loading new code while Infinitime is running. On an operating system like Linux that's not hard. You just call dlopen() and dlsym() on a file and the OS gives you references to a symbol (function, variable, etc.) in that file. FreeRTOS doesn't provide anything like this since that kind of functionality is rarely needed on embedded devices. So you'd have to build it yourself which means writing a lot of low level code to deal with pointers and memory. Somebody might put in the work to do that in the future, but its nothing you quickly put together on an evening.

CoderThomasB, 2021-06-11:
Maybe it could be an executable that when run uses an API to register the appropriate function for handling screen updates and so on. Then you woude not need to reimplement dlsym() and dlopen() you would just need to implement the API and run the executable.

Avamander, 2021-06-13:
> Maybe it could be an executable that when run uses an API to register the appropriate function for handling screen updates and so on.

Executables don't exist in the way you're imagining without a lot more OS-level abstraction what InfiniTime lacks. The OS first has to somehow execute your executable somehow while still staying running on its own, otherwise nothing's really providing the API you're imagining.

MysteriousLog6, 2021-06-14:
Cant it be a watch face resource folder inside the filesystem that is read/write so there can be .zip files that are unpacked on the phone and then sent to the watch?

Kind of similar to the mi band 5 which has a watch face file that is just a bunch of resources compressed with a info file which the OS of the watch will take and use the info file for finding the resources and VOILA! it works.

Raupinger, 2021-06-14:
from what I've seen thats basically just images to put behind the time though.

CoderThomasB, 2021-06-14:
> Cant it be a watch face resource folder inside the filesystem that is read/write so there can be .zip files that are unpacked on the phone and then sent to the watch?
>
> Kind of similar to the mi band [...]

I think using a background image and then having a config file that tells the OS where to place dynamic widgets like text, dials and analog watch would be a good idea. In addition the config could specify things like colour, font, size, line wait, font style, lengths of the watch arms on an analog watch and so on. This would be less configurable than loading code that draws the watch face to the screen but it would be much easier to implement.

JF002, 2022-08-02:
I've just created a more generic feature request about dynamic app and custom watchface to group multiple requests around that topic. Feel free to continue the discussion in #1262 ;-)

closed 2022-08-02 by JF002