InfiniTimeOrg/InfiniTime #422: Is there any way to , go back to do a rollback of the firmware incase you mess up?

MysteriousLog6, 2021-06-09:
Title.

Explanation:
Incase I were to flash an incompatible/broken firmware is the device gone? anyway to rollback if it doesn't boot?

Planning to buy one so I would like to know before hand.

Raupinger, 2021-06-09:
For sealed units flashed over DFU, the bootloader will automatically choose the last verified firmware when the watch is reset. While that has become fairly robust, it's still not impossible to brick a sealed unit, especially if you mess up flashing the bootloader itself. I would still recommend the unsealed ones for development, you can always use SWD to re-flash those.

MysteriousLog6, 2021-06-09:
> For sealed units flashed over DFU, the bootloader will automatically choose the last verified firmware when the watch is reset. While that has become fairly robust, it's still not impossible to [...]

How do you use the SWD? Do you have to do soldering? And how do you set it up?

JF002, 2021-06-10:
As @Raupinger said, the bootloader tries its best to prevent bricking : it keeps a copy of the previous firmware and can revert to this copy in the case new firmware does not run (or if the user manually requests it).
However, it depends on how bad the broken firmware is. There are edge cases the bootloader cannot support (ex : the broken firmware overwrite the memory region allocated to the bootloader, run into an infinite loop while still refreshing the watchdog,...).

To use SWD, you need a SWD debugger (STLinkV2, JLink, Black Magic Probe,...). Then you need to connect the 4 SWD pins to the PCB of the pinetime (you need to open it). For this, you can use the cables provided with the pinetime, use the pogo jig, solder the wires, a connector,...

Finally, you need a software on your computer (openOCD, for example) to flash firmware using the SWD interface.
You can have a look at the [wiki](https://wiki.pine64.org/wiki/Reprogramming_the_PineTime) and at this [doc](https://infinitime.io/blog/2020-09/i-received-my-pinetime-devkit-now-what/) for more info.

JF002, 2021-06-30:
I think we can close this issue. Feel free to reopen it if you have any other question on this topic.

closed 2021-06-30 by JF002