InfiniTimeOrg/InfiniTime #424: Feature request: Add bluetooth and charging icons to analog watchface

bertput, 2021-06-10:
The digital watchface has these icons but the analog one does not appear to have them?
Is it possible to add those icons?

rmasp98, 2021-08-02:
Should be super easy. Just add it to  InfiniTime/src/displayapp/screens/WatchFaceAnalog.cpp copying how they do it in the digital face (except different position I guess). Happy to implement once I get my dev environment up and running

Riksu9000, 2022-03-21:
The charging icon has been added to the analog watchface. Reliability of the bluetooth connection has been improved greatly recently, so I don't need to think about the bluetooth status and would rather have a simpler watchface. Do you think it would be a useful thing to add, or were you suggesting it because the digital watchface has it @bertput ?

bertput, 2022-03-21:
Hi @Riksu9000 , I use the analog watch face almost exclusively and have noticed that the bluetooth connection sometimes drops and I'm not aware of it because the bluetooth icon is not on that watchface.  I'll admit that I haven't upgraded the firmware in a while :-)
Yes @rmasp98 I could do that change myself except I don't have the development environment up on my machine yet, I thought I would be quicker to have someone who has the dev environment all set up do it. :-)

Riksu9000, 2022-03-22:
I would highly suggest you upgrade the firmware. Most, if not all issues with bluetooth have been fixed, and there are significant improvements with each release.

closed 2023-02-21 by Riksu9000