InfiniTimeOrg/InfiniTime #431: Consolidate common watch face code

jonvmey, 2021-06-13:
I'm thinking about refactoring the common parts of `WatchFaceDigital` and `WatchFaceAnalog` into a common base class. This should help reduce overall code size a little by eliminating duplicated code for similar functionality and make it easier for people to create new watch faces since they can start with some core features and not need to re-implement them themselves.

Everything related to drawing the watch face would be left up to the derived classes since that's the main thing that distinguishes them. The state management for things like date/time, battery charge level, BLE status, notifications, etc. would be provided by the base class.

Just looking for feedback on whether or not this seems worthwhile.

Avamander, 2021-06-13:
I think this would be a good approach. 

Though there's a very real chance that [PineTimeStyle](https://github.com/JF002/InfiniTime/pull/334) might be merged, so it might be a good idea to either wait for that to be merged first.

kieranc, 2021-06-14:
I agree that this is a good idea, with a view to easily swappable watchfaces in the future

riban-bw, 2024-01-17:
This may resolve issues with different behaviour between screens. Currently the analogue face does not use the StatusIcons class so it will not follow any changes made in that class. As an example, I changed the analaogue face to show a blue icon when BLE connected, grey when disconnected and hidden when disabled which only works in that face. Behaviour changes as you change views, e.g. accessing settings reverts to previous (white if connected, hidden if disconnected or disabled) behaviour.