InfiniTimeOrg/InfiniTime #435: LVGL Tick gets incremented too fast

Riksu9000, 2021-06-15:
LVGL tick tells LVGL how much time has passed.
LVGL tick is incremented in: https://github.com/JF002/InfiniTime/blob/e90e8c7426f9a44e696b208a5b393a60476a3d43/src/main.cpp#L180
According to LVGL documentation, this should happen once every millisecond at most.
https://docs.lvgl.io/latest/en/html/porting/tick.html

However, examining the code of BatteryInfo.cpp shows that even though this task is set to happen once every 1000ms, it happens much more often. This task creates the animation by adding 1% to the bar, and it gets run many times per second.
https://github.com/JF002/InfiniTime/blob/e90e8c7426f9a44e696b208a5b393a60476a3d43/src/displayapp/screens/BatteryInfo.cpp#L69

I tried to move the increment function to vApplicationTickHook(), like the LVGL documentation says, but that seemed to increment much slower. Closer to once every 10ms, making the task run once every 10 seconds.

According to this line tick rate should have been 1024Hz, but maybe this is something else.
https://github.com/JF002/InfiniTime/blob/e90e8c7426f9a44e696b208a5b393a60476a3d43/src/FreeRTOSConfig.h#L62

Avamander, 2021-06-15:
Except a tiny tiny bit faster animations, it affects what?

Riksu9000, 2021-06-15:
Well the difference is not exactly tiny. I didn't explain it very well, and I don't know any exact amounts, but the bar fills to 50% in about 2 seconds, so it runs about 25 times per second, when it technically should run just once.

If we don't know exactly how often the tick gets incremented, we can't accurately time events, and every time we use anything in LVGL that requires timing, we need to work around this issue.

On top of animations and tasks, this is probably also related to my experiments with enabling long press events on buttons, which activate practically instantly with the current setup.

JF002, 2021-06-30:
Indeed, `lv_tick_inc()` is called in the idle callback of FreeRTOS. I thought this callback would be called every 1ms as this is the tick rate of the 'tick' timer, but I've probably never checked that.

Maybe we need to find a better place to call this function so that we call it less often (and use less cpu time) ? 
DisplayApp might be a good candidate as we need to ensure there's no data race between this call and the LVGL processing done in display app.

JF002, 2021-07-13:
Fixed in #470

closed 2021-07-13 by JF002