InfiniTimeOrg/InfiniTime #445: Stopwatch+timer bug

Itai-Nelken, 2021-06-21:
Setting a timer and then starting the stopwatch causes the image to shift up when the timer opens when it ends.
It even caused a bootloop on my brother's devkit.

Riksu9000, 2021-06-21:
Can confirm, pretty spooky. Stopwatch doesn't even need to be running, the app just needs to be open.

[image]

Raupinger, 2021-06-22:
Honestly, I'm clueless on this one so far. The way the timer gets opened is mostly identical to how notification previews are. Those don't break.
I also don't see a pattern in which apps show the issue and which don't. 
Works: Clock, Screen List, Quick Settings, Steps, Meter, System Info, Display Timeouts, Wake up, Time Format
Doesn't work: Stopwatch, Music, Settings List, Heart Rate, Navigation, Paint, Pong
Additionally, some apps don't work on older branches but do on current develop.

kieranc, 2021-06-22:
I can also confirm this bug, the display remained shifted up until I rebooted.

JF002, 2021-06-30:
Wow that's not good! Is there an easy way to reproduce this issue? On which versions/commit does this happen?

hamblingreen, 2021-06-30:
Easy way to reproduce bug:
1. Start timer
2. Move to Pong app (testing what other apps do right now)
3. Wait for timer to go off
4. Bug is reproduced reliably (v1.2.0)

Itai-Nelken, 2021-07-01:
I confitm tgat this doesn't happen with the 2048 and accelerometer apps.

JF002, 2021-07-02:
@hamblingreen thanks for the procedure! I can indeed reproduce the issue!
This is kind of funny result :D We probably have a issue (data race?) around the hardware scrolling implementation (https://github.com/JF002/InfiniTime/blob/develop/src/displayapp/LittleVgl.cpp#L83)

Bloodaxe95, 2021-07-25:
Not sure if this is the right place, but this bug is still present in 1.3.0 of InfiniTime. At least I could reliably reproduce it on my sealed PineTime.

hamblingreen, 2021-08-04:
New way to reproduce something similar:
1. navigate to menu containing the Draw app
2. tap the draw icon twice quickly (this takes a few tries to get the timing down)
3. screen is slightly shifted up, and a restart fixes it.

[Video too big to upload](https://t.me/pinetime_dev/15990)

tmilburn, 2021-08-19:
This must come down to how LVGL is drawing to the screen (see https://docs.lvgl.io/master/overview/drawing.html) and the comments at #466.

I think the scrolling behaviour needs a bit of a rethink and should probably use lv_scr_load_anim https://docs.lvgl.io/latest/en/html/overview/display.html#_CPPv416lv_scr_load_animP8lv_obj_t18lv_scr_load_anim_t8uint32_t8uint32_tb

hamblingreen, 2021-09-04:
Can confirm (as expected) not fixed by 1.4

RiderExMachina, 2021-09-30:
Still an issue on 1.6.0.

Quantum-cross, 2021-10-08:
Can yall try #731 ?

SteveAmor, 2022-03-13:
Still an issue on v1.8

PR #1053 (2022-03-28): Fix display corruption when the timer is triggered
Fixes #445
This display was "corrupted" because the app do not have a background label that cover the whole screen. Because of that, LVGL would not refresh the whole display at once, but would refresh it in 3 non consecutive parts, which confused the HW scrolling transition.