InfiniTimeOrg/InfiniTime #446: Enhancement: Turn off notifications while charging

qbinary, 2021-06-21:
Maybe it's just taste, but I think it would seem more clean if vibrations would be deactivated as a whole while the pinetime is charging.

Riksu9000, 2021-06-21:
I think that showing notifications is fine, but I agree that vibrations should be disabled while charging. Is this what you meant? If that is the case could you update the title to "Turn off vibrations while charging"?

JF002, 2021-06-30:
According to what I read in some issues/PRs and also in the chat room, it seems that some people want notifications while it's charging, and some other don't want them.

Adding a configuration parameter could be a solution, but it'll also add more complexity to the code and to the UI (too many settings could confuse the users).

medeyko, 2022-02-06:
@JF002
> Adding a configuration parameter could be a solution, but it'll also add more complexity to the code and to the UI (too many settings could confuse the users).

The solution might be an "additional options" group, that will contain miniscule options like this.

Also it might be (in a perspective) a work for the companion apps.

Riksu9000, 2022-02-14:
Reducing the vibration strength while charging to reduce the awful noise may be a decent compromise here.

medeyko, 2022-02-15:
@Riksu9000:
I guess there may be quite different use cases for different people. Some would eliminate or reducing the vibration during charging in order to keep neighbors undisturbed. But other would set it on maximum to have a chance to notice alerts when the watch is not worn...