InfiniTimeOrg/InfiniTime #447: Bug: Notifications with no message show "null" as content

qbinary, 2021-06-21:
Some Apps ("Hill Climb Racing 2" in my case) send notifications with no message content. In this case infinitime show "null" as content. My proposal would be to set the title as content in that case and use the default "Notification" title. 
(I know it's easy to implement and I really would like to do it but I am still too busy AHH.)

Avamander, 2021-06-21:
Have you considered blacklisting that app on the companion app?

qbinary, 2021-06-21:
Yes. But I don't want to. Because I want the notifications to appear on my pinetime. In addition to that is it a conceptional problem which needs some kind of resolution independant of my specific case...

danielrparks, 2021-06-24:
I suspect that this is a bug in Gadgetbridge, not InfiniTime. You can get "null" from `String.valueOf((Object) null)` in Java (the language in which Gadgetbridge is written), but I don't think there's any (built-in) way to get that in C++ (the language in which InfiniTime is written).

JF002, 2021-06-30:
I've also noticed that "null" notification on an alarm notification from my Android phone via Gadgetbridge.

@danielrparks is right : InfiniTime only displays the content of the notification sent by the companion app. In this case, the companion app actually sent that "null" message. I would suggest to fill a bug report on your companion app project. You can even post the link in this post so that we can follow it ;)

marbetschar, 2021-08-21:
I do have the same issue and opened a corresponding issue at Gadgetbridge's Issue Tracker: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2373

closed 2021-10-02 by Avamander