InfiniTimeOrg/InfiniTime #460: Music controls don't work for iOS

zac-garby, 2021-06-26:
The music app can't control the iOS Music app. I'm assuming this is just not implemented (yet?), but I thought I'd make an issue in case I'm doing something wrong or I need to set something up in order for it to work.

Otherwise, if it's not implemented yet, are there plans for iOS music controls in the future?

thefenriswolf, 2021-06-26:
Which companion app are you using?

Avamander, 2021-06-26:
BLE standard doesn't contain one for music control (BT Classic does, but we don't have that), such functionality has to be implemented by the companion app.

zac-garby, 2021-06-26:
ah that makes sense. i'm using nRF Connect, but only because afaik it's the only one for iOS

JF002, 2021-06-30:
Indeed, NRFConnect is a generic app that allows to connect to devices, browse their characteristic, debug the connection,... but it does not provide any higher level functionalities like music control, navigation or notifications.
Unfortunately, I think there's no companion app compatible with PineTime / InfiniTime on iOS for now. Someone want to fill that gap?

zac-garby, 2021-06-30:
if i had more time i'd definitely look into making one. also that explains what the notifications section is for, i take it that the watch can show your phone notifications? if so, i'm not sure if iOS actually allows apps to look at systemwide notifications, i would guess not?

Avamander, 2021-06-30:
> i'm not sure if iOS actually allows apps to look at systemwide notifications, i would guess not?

You can maybe find the Pebble app's App Store description and see if it implemented it. It might actually be doable, but I haven't personally paid attention.

JF002, 2021-07-02:
I would be surprised if iOS doesn't allow notifications to be sent to any smartwatch, but I don't know much about the Apple world so...

zac-garby, 2021-07-02:
yeah looks like it's possible actually, i was just thinking how obsessed apple is with privacy recently

tapresle, 2021-07-11:
Pebble used to do notifications on the original Pebble, I never had a Pebble Time or the others so I can't speak to the other functionality. I've actually recently received my InfiniTime and made a quick and dirty app to set the time via iOS. I'll see what's involved in getting notifications and basic music controls working but can't guarantee any timeline.

tapresle, 2021-07-12:
I've done some more digging and it looks like [Apple Media Service](https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleMediaService_Reference/Specification/Specification.html#//apple_ref/doc/uid/TP40014716-CH1-SW18) needs to be implemented on the InfiniTime firmware. Similarly, [ANCS](https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Specification/Specification.html#//apple_ref/doc/uid/TP40013460-CH1-SW7) for notifications. If there's someone more comfortable with editing the firmware to implement those I'd be happy to test, I don't quite have the tools I need to edit firmware yet and it's been almost a decade since I've looked at C/C++ code 😆

computergeek125, 2021-11-24:
Was reminded this issue exists today - I have an iPhone, an unsealed dev watch, a Mac, and probably enough Unix/Linux machines to figure out how to compile Infinitime. I can also help test.

TheAwesome98-Real, 2021-12-30:
Old issue but InfiniLink works, but only with Apple Music so far. There was an issue saying system wide media support is coming soon iirc

JF002, 2022-01-01:
Thanks for the info! My understanding is that it works only with the Apple Music app, and this is a limitation of iOS. 
Other apps would need other BLE services and characteristics to be implemented in InfiniTime.

tapresle, 2022-01-01:
Not sure why this was closed but that is correct, as I put in an earlier comment, Apple Media Services would need to be implemented in the Infinitime ble stack itself. I’d help out but unfortunately don’t have the skill set or tools to write and flash code directly to the device.

----------------------------------
Ted Presley
***@***.***
Mobile: (336) 816-2053
tedpresley.com

JF002, 2022-01-01:
I closed it because there's nothing we can do in InfiniTime to make iOS compatible with the music control API implemented in InfiniTime.

Ok, I thought this was a "bug" issue, but, as it is a "feature request", I guess I can re-open it, in case  someone is willing to implement whatever is necessary for iOS to work correctly :-)

tapresle, 2022-01-01:
Implementing the Apple Media Service Bluetooth service is the fix here, it’s what automobile, speaker, etc use after binding to an iOS device. BangleJS which is a similar open wearable has implemented it as well. I’ll see if I can dig in sometime soon with my dev kit after I get what I need to flash the firmware, but it may take a while to get in as I get to speed and as I also need approval for any open source contributions from my employer.

----------------------------------
Ted Presley
***@***.***
Mobile: (336) 816-2053
tedpresley.com

JF002, 2022-01-02:
I think most of the active developers in InfiniTime do not own an Apple device, and that's probably the main reason why no one has already tried to implement the apple specific protocole.

That would be awesome if you could help on that topic! Feel free to ask for help here on in the community chat rooms if needed :)

jrmolin, 2022-04-12:
I have a PineTime, a lot of C/C++ experience, a fair number of Apple products, a lot of GNU/Linux experience, and some free time. I would be happy to look into this, but I am a little hazy on this code base. Could someone point me at the specifications I would need to hit for this to work?

minacode, 2022-07-02:
Hi, I tried implementing ANCS once. It seems that AMS has the same implementation issue:

> Due to the nature of iOS, the AMS is not guaranteed to always be present. As a result, the MR should look for and subscribe to the Service Changed characteristic of the GATT service in order to monitor the potential publishing and unpublishing of the AMS at any time. 

Right now, Infinitime does not have support for BLE services that can appear and disappear at any time. Sadly, Apple's Services build heavily on this mechanic. 
Thus, implementing the GATT service would be a good start. 
Sadly, I am writing my thesis and don't have time for this right now.

With ANCS and AMS running, the watch would have full support for iOS for everything but navigation, I assume.

cyberneel, 2025-01-07:
I got ANCS working and opened a PR recently, #2217. The watch seems to be able to reconnect to te service on its own and everything works. I think I may be able to implement AMS as well.