InfiniTimeOrg/InfiniTime #461: Request: Extended symbols support

JanezCZ, 2021-07-01:
When using InfiniTime (paired with GadgetBridge) with other languages, there can be difficulties reading notifications.
For example czech language uses extended latin and letters in notifications like _č, ž, ř, á_ and so on are not displayed correctly. Some of them replaced by space, some of them are just ignored.
This is just one example but I believe there will be more languages with similar difficulties.

Example:

- _Zpráva_ -> Zprva
- _kód_ -> kd

Avamander, 2021-07-01:
We don't have the internal storage for a lot of symbols, so symbols missing are simply not supported.

There's two possible solutions, either GB should transliterate or external storage should contain a font that has a wider range of symbols.

JanezCZ, 2021-07-02:
I tried to make some research. Looks like both fonts I found in displayapp (JetBrainsMono-Bold and open_sans_light) already support extended latin symbols.
Transliteration in GB is specific by device since 0.56.1 so I will try to request adding this feature for InfiniTime as well. However I hope it is temporary solution even though I'm unable to contribute to code on my own.

JF002, 2021-07-02:
As @Avamander said, the internal storage of the PineTime is kind of small and there's not enough room to store all symbols from all languages.
In the future, we might be able to use the external memory to upload and store fonts with more (language specific) symbols.

calhix, 2021-09-04:
> As @Avamander said, the internal storage of the PineTime is kind of small and there's not enough room to store all symbols from all languages.
> In the future, we might be able to use the external memory to upload and store fonts with more (language specific) symbols.

If I understand correctly, could we perhaps in the future be able to store fonts in the external memory just for the languages one would want to avoid potential storage issues? It's unfortunate for asian speakers (and I assume all non-latin languages) since every message notification is blank at the moment.

moriel5, 2021-09-05:
While I personally don't really use the notifications feature, this does but a stop for me being able to recommend the PineTime to (knowledgeable) friends who are not English-speakers (we use an RTL language here), so I agree, we need to find a solution to this (storing fully in the external storage sounds nice, however we'll need to create new fonts that are light enough).

Avamander, 2021-10-02:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/212

closed 2021-10-02 by Avamander