InfiniTimeOrg/InfiniTime #462: Settings list code isn't dynamic.

Riksu9000, 2021-07-01:
https://github.com/JF002/InfiniTime/blob/develop/src/displayapp/screens/settings/Settings.cpp

In Settings.cpp, there are separate arrays of applications for each screen. This is easy to fix by creating one array of all apps in Settings.h and replacing the old CreateScreenN functions with a single new function that uses this array.

The real issue is when we add a new app and have to create a new screen. This always requires changes in multiple places in the code. It would be better if the code could automatically create enough screens for all apps. Then it would be much easier to add, remove and move around options and maybe even change them at runtime.

I wanted to add a debug menu to the settings for developers. This menu would have to be hidden somehow for non developers. This is much harder to do with the current code reasonably.

All of this applies to ApplicationList as well, so that can also benefit from solving this.

JF002, 2021-07-02:
The idea behind this implementation is to be simple, and to avoid memory allocation at runtime (dynamic memory allocation). But I agree with you, when you want to add a new app/setting app, you have to manually add the new app , the CreateScreenN function, increase the size of the array, add the symbol in the font, add the app in Apps.h,...

This could certainly be refactored and improved, indeed.

Avamander, 2021-10-02:
If this is still something that really needs attention, let's reopen.

closed 2021-10-02 by Avamander

Riksu9000, 2021-10-02:
I think it does if we want to separate apps from the firmware. The code would need to be able to dynamically create enough screens to fit any number of apps. Updated the title as this applies to applist as well.

Avamander, 2021-10-02:
Let's create a clean new issue that lists the potential steps towards fully dynamic watchapps?

In theory you could create a dynamic app menu that's separate from the system's built-ins, which sounds reasonable.

JF002, 2021-10-02:
If this is a subject some contributors want to work on, I can create a "project" dedicated to the implementation of dynamic apps. That would give a space to think all the steps needed to reach that goal. What do you think?