InfiniTimeOrg/InfiniTime #463: Step Count Impossibly High

hamblingreen, 2021-07-02:
Specs: Sealed PineTime running v1.2.0 connected to an Alcatel A502DL over Gadgetbridge.

Issue: Step count is at 536876472 in both the watch face and steps "app". (For reference I went to bed with it at around 10000 and woke up to this) On the watch face, not in the app, it flickers between this number and 0 repeatedly. It will not count any higher than this number.

How to reproduce: Unknown

Issue is resolved when restarting watch.

JF002, 2021-07-03:
From a software point of view, it looks like it's reading garbage from the sensor. This could be caused by a bad connection between the sensor and the mcu, a faulty sensor or... an unknown software bug.

As we talked about this in the chat, have you tried to reset (long push on the button) and / or wait for the battery to drain completely to force a hardware reset?

hamblingreen, 2021-07-04:
Yes I have tried both of those as well as re-flashing firmware. I'm putting off breaking the thing open because of the hassle it'd be to reseal it. Also I only have e-6000 not b-6000 so I would not be able to use the same stuff as in the factory.

JF002, 2021-07-04:
I have only noticed that kind of behavior during development. I guess breakpoints, crashes and resets cause bus errors. In this case, I disconnect power (open unit), wait a few second and start again.

Maybe wait a bit longer when the battery drains, to ensure that voltage drops low enough to reset the accelerometer?

hamblingreen, 2021-07-08:
Alas, she did not read from the accelerometer after 24 hours with the battery fully drained. I don't know what else to say other than that the connection from the sensor and the mcu must've been severed at some point after I got the thing. This issue I opened seems to be directly connected to a hardware issue and not the firmware, so I should promably close this issue. Feel free to contact me on telegram if you have any further suggestions. Thanks.

closed 2021-07-08 by hamblingreen