InfiniTimeOrg/InfiniTime #486: Companion app for iOS

JF002, 2021-07-13:
Recently, we've seen more and more users and potential users asking for a companion app running on Apple devices and more precisely on iOS.
While we already have a companion app for Android ([Gadgetbridge](https://gadgetbridge.org/)), and 2 for Linux ([Amazfish](https://github.com/piggz/harbour-amazfish) and [Siglo](https://github.com/alexr4535/siglo), we have no app for the Apple world.

If anyone is willing to either port an existing app to iOS/Apple or create a new app, know that your contributions will most likely be well received by the community ! 

Don't know where to start? Here are some ideas of functionalities:
 - Discover and connect to a PineTime running InfiniTime
 - Request basic info (firmware version, hardware version, manufacturer)
 - Set the date/time
 - Display the battery level, display an alert if the battery is low, keep an historic of the battery level
 - Update the firmware over-the-air using the DFU protocol
 - Send notification to the PineTime (messages, calls,...)
 - Bridge with a navigation app
 - Music control
 - Heart rate monitoring
 - Support other firmwares (like [WaspOS](https://github.com/daniel-thompson/wasp-os) and [PineTimeLite](https://github.com/joaquimorg/PinetimeLite))
 - ...

Of course, these are just ideas, and not all of them are needed to make your future app enjoyable!

Do not let yourself discouraged by the cost of an Apple developer account, Pine64 will probably be able to arrange something for you :-)

gamelaster, 2021-07-18:
iOS Companion app would be amazing! Also, I think the app can be made for example in Flutter, so it will even work on Android or PC.

(BTW About the Apple Developer Account license, I plan to purchase one for myself, so if there will be need for dev account, I can publish it 😊)

markyjackson-taulia, 2021-07-18:
I am going to try to see if I can make this work

moriel5, 2021-07-20:
@gamelaster Please, just not Flutter.

While it makes porting easier, the performance is amongst the worst I have seen (tested on low and high performance devices, on both Arm (including Arm64) and x86 (including x86-64), on Windows, Linux, and Android).

gamelaster, 2021-07-20:
@moriel5 It was just idea, but I checked some benchmarks, and of course it will be not that fast as native, although I think it's performant (and from what I've seen it's faster than for example React Native).

Anyway, when I think about it now, it can be more problematic, since hard to say if Bluetooth LE library for Flutter will have all required features, also if there are libraries to work with iOS notifications etc.

fbanning, 2021-07-20:
This may be totally off but it could the [Kirigami framework](https://develop.kde.org/frameworks/kirigami/) be a good fit? I'm not 100% sure it will work on iOS but the landing page states that apps have been built for it. Big up-side would be that code could be written in C++/QML and that a single app could be used on all major platforms (streamlining dev efforts).

Avamander, 2021-07-20:
Very little of the overall complexity is in the end in the UI alone. Reuse will be somewhat minimal. Most of it stems from what is platform-specific, like stable Bluetooth, reliable background services and integration with platform features. 

Worth trying but don't underestimate the work required.

Avamander, 2021-07-20:
@gamelaster 

> Anyway, when I think about it now, it can be more problematic, since hard to say if Bluetooth LE library for Flutter will have all required features, also if there are libraries to work with iOS notifications etc.

If you have characteristic write, read and notify then that's all you're going to need.

JF002, 2021-07-20:
@fbanning I think [Amazfish](https://github.com/piggz/harbour-amazfish) is based on Kirigami, and it already supports InfiniTime!

laubblaeser, 2021-07-21:
> 
> 
> @fbanning I think [Amazfish](https://github.com/piggz/harbour-amazfish) is based on Kirigami, and it already supports InfiniTime!

Cool, thanks for the info. I knew that it has always been a Qt app (coming from SFOS) but didn't know that they've been using Kirigami.  
Maybe it's worth the effort to ask over there and see if it's possible or has been considered before. (I just did that with my private account since I just realised I'm using my work account here. 😅)

ocdtrekkie, 2021-08-02:
Just got my PineTime, and a little shocked I can't even set the time on it without an Android or Linux device... so yeah, this would be really nice.

jbkim, 2021-08-03:
@ocdtrekkie I found this https://hubmartin.github.io/WebBLEWatch/ and set the time.

ocdtrekkie, 2021-08-03:
@jbkim Possibly helpful, but I make a point not to touch anything Google with a ten foot pole. The requirement on a Chrome-proprietary makes it as much use to me as an Android app. :/ I have some Linux devices, but apparently neither of those companion apps "just install" on my Pinebook Pro. (And sent me down another rabbit hole trying to figure out what I need to upgrade my Pinebook Pro to a supported OS... but I digress...)

I know this is experimental hardware with experimental software, but I feel like being able to use a watch to tell the time should be the first thing that "just works" without external hardware.

Looks like Chrome's Bluetooth support may have made it into Edge, which my Windows machines are burdened with... maybe your solution will work for me there. Thanks, I'll keep hacking on the problem.

ocdtrekkie, 2021-08-03:
@jbkim Update: After finding a Windows device where I don't have that browser flag disabled by policy, that does have a Bluetooth radio... I have in fact, set the time on my watch. You're my hero. Thank you.

ocdtrekkie, 2021-08-04:
@xan-m That's really exciting. I'd be happy to test/give feedback once you have something distributable!

xan-m, 2021-08-06:
I've got a repo up and running for my iOS companion app prototype! So far, I can:
- Set the time automatically on connecting to a PineTime
- Read heart rate and battery from PineTime, and display data from both real-time in the app
- Read button presses from the PineTime the music app - just printing results to console for now, but it's the first step to interacting with the phone's music controls
- Send test notifications from the app to the PineTime - just a debug button for now to send a test notification, but it's the first step for notifications too!

Please come check it out! As I said above, I've never worked with this language or these technologies before, so please excuse any poorly-written code. If you have a Mac and are willing to try building the app to test it on your local machine, I'd really appreciate it. If not, it'll probably be a while before I'm ready to upload it to TestFlight so I can distribute betas, so hang in there!

I'm also 100% open to changing pretty much any part of this project, I just wanted to get the ball rolling on something. If you have a better idea for the name, feature suggestions, licensing suggestions, etc. please let me know!

https://github.com/xan-m/Infini-iOS

ocdtrekkie, 2021-08-06:
Awesome, @xan-m. I don't have a Mac, so I'm probably holding for TestFlight or something of that nature, if you plan to publish, but I will take a look at the project a bit when I can.

MattiaFochesato, 2021-08-16:
In order to implement music controls and notifications / calls on iOS, firmware support must be added. Apps on iOS don't have permissions to read notifications and to control music. Furthermore, it is not possible to be sure that the app is always open in the background.

It is necessary to implement in the firmware the specific functions that Apple makes available for BLE devices.

One method that could be used is that when the device is paired, it is told whether this is an iPhone or not. If it is an iPhone, the various controls are activated.

- The Apple Media Service (Playback info and control): https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleMediaService_Reference/Specification/Specification.html
- Calls and Notifications: I think (but i'm not sure) that the watch should act as a HID device.

xan-m, 2021-08-16:
Yep, we discussed this the other day in the PineTime chat. ANCS is definitely required for notifications and AMS is required for system-wide music controls. The main issue with ANCS right now is that it requires secure pairing. From what I can tell from the chat, secure pairing is not a priority until the current BLE connection stability issues are ironed out (please correct me if I'm wrong here!).

My dev kit is out for delivery as we speak, so I'll try to play around with AMS and ANCS implementations in the coming weeks.

xan-m, 2021-08-22:
Infini-iOS is now available on TestFlight! iOS users, come grab it and let me know what you think!

https://testflight.apple.com/join/Z7u1Jxp4

ajack2001my, 2021-09-06:
Sadly, I only have an iPhone 6 with iOS 12.x installed... :(