InfiniTimeOrg/InfiniTime #490: Question: Pulling Activity Data to Computer

atomdmac, 2021-07-14:
My apologies if there's a better place to ask this question.

It's currently possible to view accelerometer data in real time directly on the pine time.  However since the screen is so small it's only possible to display a limited amount of data at a time. This makes it difficult to observe patterns when certain motions are performed (ex raising and lowering one's hand).  

I'd like to be able to export this data to my computer so that I can graph it and view larger spans of time. Is this currently possible with something like gadget bridge? I was able to export data from gadget bridge but it doesn't seem to contain any of the accelerometer data.

For what it's worth I want to get this information so that I can help make improvements to the raise to wake functionality.  (Specifically, I'd like to make it less sensitive and also allow the screen to turn off when I lower my hand).

If there's a better place to ask these types of development questions please let me know. Thank you!

JF002, 2021-07-14:
This is the good place to ask this question :+1: 

There's currently no way to store and export data those data to the companion apps, except for the current HR value, as you mentioned.

However, we would like to work on features like this, especially to expose fitness data (hr/motion/steps) to companion apps. I created a Github project for this : https://github.com/JF002/InfiniTime/projects/4

We'll need to store the data to send them later on to the companion app. For this, we need an easy way to store those data. Fortunately, the first steps toward that goal have been recently reached : we have a file system in the external flash memory. We are now able to easily create a "file" in the watch memory to write any data we want! This is part of this project : https://github.com/JF002/InfiniTime/projects/3

So, yes, your request is totally reasonable and doable. Now, we need to actually do it ;)

atomdmac, 2021-07-14:
Thanks, @JF002 !  I'll keep an eye out.

Riksu9000, 2021-07-18:
It looks like the chip has built in detection for wrist tilting. Any reason not to use this?

https://github.com/JF002/InfiniTime/blob/57b339707861c5688f5d432f1506a99df6bb0fce/src/drivers/Bma421_C/bma423.c#L1979

"Wake up display on wrist tilt"
https://github.com/BoschSensortec/BMA423-Sensor-API

atomdmac, 2021-07-18:
@Riksu9000 My intention (at least in the longer term) is two fold:
1) Refine the conditional checks for when the display should wake (ex. the wake sometimes responds to "false positives")
2) Allow the display to turn off again when I put my hand back down

My thought was that if I had the ability to study the data generated by performing certain actions, I would be in a better position to tackle the items in the list above.  The graph app on the watch is great for getting a general idea of how the accelerometer "sees" certain actions but the data go by too quickly to get a good look.

That said, I have limited experience with `C++` and none at all with embedded systems so it's very possible that I'm taking a naive approach here.

Riksu9000, 2021-07-18:
>Refine the conditional checks for when the display should wake (ex. the wake sometimes responds to "false positives")

It is true that by creating the checks manually we get more control, however getting it to work well might be difficult. We also don't know how accurate the builtin detection is before we try it, or maybe someone already has and I just don't know.

>Allow the display to turn off again when I put my hand back down

This it probably can't do, however automatically turning off the display would be a new source of possible false positives, so any small gains from this might not be worth the potential issues.

I might even be wrong about this built in feature, but I think it could be worth looking into still.

---

If we are going to refine the detection, we should also consider if it would be worth it to change the configuration and improve communication with the sensor:

- The sensor supports interrupts which could be used to read the data immediately as it's available.
- In [Bma421.cpp](https://github.com/JF002/InfiniTime/blob/develop/src/drivers/Bma421.cpp) the chip has been configured to refresh the data at 100Hz and limit the range to 2G, but systemTask is currently polling the data at just ~10Hz.

atomdmac, 2021-07-19:
> It is true that by creating the checks manually we get more control, however getting it to work well might be difficult. We also don't know how accurate the builtin detection is before we try it, or maybe someone already has and I just don't know.

I believe @JF002 implemented the [current raise-to-wake functionality](https://github.com/JF002/InfiniTime/pull/255).   He may know if there's any dis/advantage to using the chip-provided RtW facilities.

> I might even be wrong about this built in feature, but I think it could be worth looking into still.

I think I had initially misunderstood what you were proposing (I didn't realize that the current raise-to-wake implementation in InfiniTime was separate from the one provided by the chip).  What you're saying makes sense to me (for what it's worth :P ).  It would definitely be interesting to see how the two implementations compare in practice.

> This it probably can't do, however automatically turning off the display would be a new source of possible false positives, so any small gains from this might not be worth the potential issues.

:point_up: This is a fair point and is the main reason that my initial question was around what capabilities currently exist for recording/gathering sensor data.  It's very possible that the two actions (i.e. raising and lowering the wrist) don't have different enough "signatures" to be able to trigger wake/sleep reliably.  I figured the fastest way to see how different the two actions appear from the point of view of the sensor would be to get access to the data in a form that is easier to inspect.

Avamander, 2021-10-02:
I think the question has been answered.

closed 2021-10-02 by Avamander