InfiniTimeOrg/InfiniTime #493: Bug/Feature: Persist Step Count Between Resets

atomdmac, 2021-07-17:
I've had to reset my watch about once a day for the last week or so (usually to fix random Bluetooth connectivity issues).  When this happens, my daily step count is reset.  It would be really nice if it were possible to persist the step count across resets.

JF002, 2021-07-18:
Yes, right now, the step counter is re-initialized on every reset, which also reset the step count.
One way to achieve this would be to probe the bma42x on reset to determine if it's already initialized or not. In theory, the fw should have to init the chip only on cold boot (due to power loss). On reset, the chip should be alreay initialized.

Another way to do it would be to store the step count in persistent storage, now that we have littlefs integrated and use this value as an offset to the value measured by the sensor.

Avamander, 2021-07-20:
Related to https://github.com/JF002/InfiniTime/issues/321