InfiniTimeOrg/InfiniTime #499: MIDI Player

moriel5, 2021-07-20:
It would be nice to be able to use the PineTime with Bluetooth headphones (or speakers) to playback MIDI files (especially useful for people like me, who play musical instruments).

Avamander, 2021-07-20:
Just like with #500, audio use-cases use Bluetooth Classic, we don't and can't support that any time soon.

moriel5, 2021-07-20:
Understood, hopefully it will be a possibility in the future.

JF002, 2021-07-20:
What kind of use case do you have in mind (knowing we have only 512KB + 4MB of storage for the whole system) ?

moriel5, 2021-07-20:
Mostly for playback of sheet music (well, MIDI aren't actually audio files) to aid in learning said piece of music.
That way, there wouldn't need to be too many, and even then, it should be enough for a few tens of files.

moriel5, 2021-08-04:
Update: Regardless of the decisions, I can say that I have found a soundfont that works reasonably well and weighs about 1MB (still looking for something lighter which still works well) on the Internet Archive.

It's called AweROMGM, and it weighs about 1.04MB.
I forget whether it only contains sounds for the piano, or if there are a few other instruments (I found quite a few ~30-300KB soundfonts that contained sounds for several different guitars, but no pianos).

JF002, 2021-12-30:
I'm closing this feature request right now as this kind of use-case/feature is not in our scope right now.

closed 2021-12-30 by JF002