InfiniTimeOrg/InfiniTime #500: [Feature Request] Bluetooth Audio

moriel5, 2021-07-20:
Required for issue #499.

I guess Bluetooth audio isn't yet implemented in InfiniTime, as people wouldn't usually think of using something with such constrained storage and no built-in speakers for playing back audio.

Riksu9000, 2021-07-20:
This is very closely related to #499 as I don't see any other use for bluetooth audio other than for music.

This is an interesting idea, however unfortunately there is only 4MB of usable storage on the Pinetime. This is enough to store just 4 minutes of 128kb/s audio, assuming we have the entire 4MB to use, and without considering the large amount of new code that would be required to handle audio files and bluetooth speakers.

MIDI is even more complicated, requiring even more work to get working, and the samples would still probably use a lot of space.

Avamander, 2021-07-20:
Audio use-cases use Bluetooth Classic, we don't and can't support that any time soon.

moriel5, 2021-07-20:
Yeah, I did mention it that issue what would probably be the main uses for this.

Certainly understandable, however it may be possible find an audio sample with a low size (the old Nokia Classic 2600-C2 supported MIDI back in the day).

JF002, 2021-07-20:
Same question than for the MIDI player : what kind of use case do you have in mind? I mean, we could probably store a short audio sample and play it back on some BLE audio protocol, that would be a nice demo, but what would you do with it?

moriel5, 2021-07-20:
@JF002 As mentioned, MIDI playback (more information in aforementioned issue), however also hearing aids could benefit from this (like for a possible alarm clock).

blaueente, 2021-07-22:
> Audio use-cases use Bluetooth Classic, we don't and can't support that any time soon.
Just for the record: There is LE Audio now. Not that I see any chance of it being useful in InfiniTime with the current hardware.
https://www.bluetooth.com/learn-about-bluetooth/recent-enhancements/le-audio/

Avamander, 2021-07-22:
@blaueente yes, that was mentioned in #501. There basically aren't any transmitters or receivers out there for it to be useful, even if implemented. (Which it can't be without BLE 5.2 and bunch of extra work)

moriel5, 2021-07-22:
@Avamander As someone who has next to no idea how Bluetooth hardware is implemented, can't just any Bluetooth audio device (supporting BLE) and it's antennaes be a receiver, with the Pinetime's SoC and it's antennaes be a transmitter?

Avamander, 2021-07-22:
@moriel5 It's not that the hardware can't, it's that you can't go to a store and buy a speaker that could receive the audio you're transmitting (that is of course after you've gotten InfiniTime on a BLE 5.2 compatible stack and implemented BLE Audio on top).

moriel5, 2021-07-24:
@Avamander Would that be because the majority still rely on BT4.x?
Since in that case, I guess as time goes on, we'll see more that support BLE Audio (I believe that quite a few TWS headphones with Qualcomm's newest chipsets support BLE Audio, after all).

Avamander, 2021-07-24:
> Would that be because the majority still rely on BT4.x?
> Since in that case, I guess as time goes on, we'll see more that support BLE Audio (I believe that quite a few TWS headphones with Qualcomm's newest chipsets support BLE Audio, after all).

Not really that people rely on 4.x, but it is very new in terms of tech/protocol. Next gen phones will probably bring am increase of BLE Audio compatible devices.

moriel5, 2021-07-24:
@Avamander Then it is just a matter of time before it becomes relevant.

JF002, 2021-12-30:
I'm closing this feature request right now as this kind of use-case/feature is not in our scope right now.

closed 2021-12-30 by JF002