InfiniTimeOrg/InfiniTime #501: [Feature Request] Bluetooth 5.2

moriel5, 2021-07-20:
Is the Bluetooth stack in InfiniTime still at Bluetooth 5, since 5.1 and 5.2 are already available for the nRF52832 via the proprietary SDK.

It would appear that while not required, upgrading to a newer Bluetooth revision could help in regards to power savings with Bluetooth Audio (issue #500), better support for hearing aids (could be a good feature request in the future) and location/tracking.

Avamander, 2021-07-20:
NimBLE supports Bluetooth 5.1, switching to the proprietary stack is not going to happen any time soon.

Newer Bluetooth revision wouldn't help with audio, LE Audio is a separate standard that would require specific support.

moriel5, 2021-07-20:
Yeah, I never meant switching to the proprietary stack, I only brought it as an example that it is possible.

Interesting, Wikipedia explicitly mentions power savings on LE Audio in conjunction with Bluetooth 5.2.

Avamander, 2021-07-20:
LE Audio itself provides power savings over BT Classic in some cases, but it's not inherit to Bluetooth 5.2 it relies on it.

moriel5, 2021-07-20:
Ah, thanks for the clarification.

JF002, 2021-12-30:
I guess the update to an eventual newer version of bluetooth will come with newer versions of NimBLE.

closed 2021-12-30 by JF002