InfiniTimeOrg/InfiniTime #502: Feature Request: Authentication/Authorization

GIJack, 2021-07-20:
Ask for authorization every time a new phone or other device is connected. Remember authorized devices

Add menu item in settings for managing authorized devices in settings.

The easy way to do this, would be just to remember bluetooth addresses. The hardway is using crypto.

Avamander, 2021-07-20:
NimBLE supports proper bonding, there's absolutely no good reason to reinvent that with anything homebrew, it'd be only pain and misery. With filesystem support the keys created during bonding could be stored properly and kept after a reboot.

GIJack, 2021-07-20:
> NimBLE supports proper bonding, there's absolutely no good reason to reinvent that with anything homebrew, it'd be only pain and misery. With filesystem support the keys created during bonding could be stored properly and kept after a reboot.

I had no idea that it did, but even better. Lets go with that then.

PR #796 (2021-12-09): Connect and bond with a passkey
Add Feature - Connect and bond with a passkey

- [X] Passkey pairing - passkey is displayed on watch and entered on phone
- [X] Swipe down to clear passkey screen
- [X] Connection encryption
- [X] Connection bonding
- [X] Automatic reconnects to bonded peripheral\*
- [X] Trusted device on Android
- [X] Persist bond between reboots

\* ~~**Rebooting the watch will cause reconnect failures. You must delete the bond from the phone to reconnect/pair.** Persisting the bond between reboots will resolve this.~~

Closes #224 
Closes #302 
Closes #502 

There are several [... cut]
closed 2021-12-09 by JF002