InfiniTimeOrg/InfiniTime #503: Battery Life

moriel5, 2021-07-20:
My PineTime is one of the original units that originally came with the closed-source DaFit firmware, and on that former, I used to get about 2-2.5 weeks of battery life (depending on how much I interacted with it).

On InfiniTime 1.2.0, I usually get about 5 days of battery life, however.
It takes about the same amount of time to charge as it used to, perhaps slightly longer (up to 30 minutes more).

My usage is the same between them, lowest brightness setting, motion sensing to turn on the display set to off (although I have enabled DT2W on InfiniTime), and it is usually disconnected from other devices.
I usually check the time at irregular intervals, which means that on average, I check the time up to 5 times an hour, but most of the time, it is once or twice every few hours, and I have display timeout set to 5 seconds (some of the time, I manually turn off the display).

I had meant to flash pre-alpha versions of InfiniTime back then, however was met with multiple hurdles (no SWD programmer, lack of a display for my Raspberry Pi 3B, several hardware failures, which effectively meant I was without a computer some of the time), which is why I only got around to it last week (via BLE, from a phone that I still need to flash a custom ROM on before it becomes my daily driver).

JF002, 2021-07-20:
The battery life you mention is similar to what I observe on my units (around 7 days idle, 3-5 days under normal usage, ble, notifications, wake up on wrist rotation,...).

There are probably a lot of optimizations and fine-tuning to be done in InfiniTime to increase the battery life (we know it's possible, but we have to find out how ;) ). You can have a look at this thread to have an overview of the measurements and analysis we have already done : https://github.com/JF002/InfiniTime/issues/53.

moriel5, 2021-07-20:
@JF002 Thanks, I'll take a look, and try to think of suggestions (though I am very underqualified for the task).

closed 2021-10-02 by Avamander