InfiniTimeOrg/InfiniTime #511: Not able to compile on Windows 10

tlouwers, 2021-07-22:
Using commit https://github.com/JF002/InfiniTime/commit/57b339707861c5688f5d432f1506a99df6bb0fce

I try to build the sources on Windows, using the tools mentioned in the https://github.com/JF002/InfiniTime/blob/develop/doc/buildAndProgram.md guide and using Visual Studio Code and Ninja for this. The ARM toolchain is installed and I am able to use it for a pet project, an SMT32F4-Discovery board.

Just to be sure: the PATH variable is set:
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin

I have the following tasks.json file:
`{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Configure - PineTime [Debug]",
            "type": "shell",
            "command": "CMake",
            "args": [
                "-B./build",
                "-GNinja",
                "-DCMAKE_BUILD_TYPE=Debug",
                "-DUSE_OPENOCD=1",
                "-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major",
                "-DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345",                
                "-DBUILD_DFU=1"                
            ],
            "problemMatcher": []
        },
        {
            "label": "Configure - PineTime [Release]",
            "type": "shell",
            "command": "CMake",
            "args": [
                "-B./build",
                "-GNinja",
                "-DCMAKE_BUILD_TYPE=Release",
                "-DUSE_OPENOCD=1",
                "-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major",
                "-DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345",                
                "-DBUILD_DFU=1"
            ],
            "problemMatcher": []
        },
        {
            "label": "Build - PineTime",
            "type": "shell",
            "options": {
                "cwd": "${workspaceFolder}/build"
            },
            "command": "Ninja",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
    ]
}`

I delete the 'build' folder before start, then try to configure the project. This results in an error:
`> Executing task: CMake -B./build -GNinja -DCMAKE_BUILD_TYPE=Debug -DUSE_OPENOCD=1 '-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major' -DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 <

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment    
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Git/PineTime/build/CMakeFiles/CMakeOutput.log".
See also "C:/Git/PineTime/build/CMakeFiles/CMakeError.log".
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command CMake -B./build -GNinja -DCMAKE_BUILD_TYPE=Debug -DUSE_OPENOCD=1 '-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major' -DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

Avamander, 2021-07-22:
Try setting CMAKE_C_COMPILER (and CXX) manually and see if that helps. But it might also be that the cmake invocation is incorrect on Windows.

You can also switch to compiling in WSL2 where it's known to work.

tlouwers, 2021-07-22:
I tried forcing the toolchain in the tasks.json file, but that did not work, same output:
     "-DCMAKE_TOOLCHAIN=C:/Git/PineTime/cmake-nRF5x/arm-gcc-toolchain.cmake",

After that I tried to force the CMAKE_C_COMPILER:
     "-DCMAKE_C_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe",
     "-DCMAKE_CXX_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe",
which gave errors that it could not compile a dummy file to check if the compiler worked. I could get around this by adding:
     "-DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY",
after which compiling worked - now linking fails:

`> Executing task: CMake -B./build -GNinja -DCMAKE_BUILD_TYPE=Debug '-DCMAKE_C_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe' '-DCMAKE_CXX_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe' -DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY -DUSE_OPENOCD=1 '-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major' -DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 <

-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0
BUILD CONFIGURATION
-------------------
    * Version : 1.2.0
    * Toolchain : C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major
    * GitRef(S) : 57b3397
    * NRF52 SDK : C:/Git/nRF5_SDK_15.3.0_59ac345
    * Programmer/debugger : OpenOCD Client
    * Debug pins : Disabled
    * Build DFU (using adafruit-nrfutil) : Enabled
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Git/PineTime/build

Terminal will be reused by tasks, press any key to close it.

> Executing task: Ninja <

[63/527] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/screens/SystemInfo.cpp.obj
C:/Git/PineTime/src/displayapp/screens/SystemInfo.cpp: In function 'const char* {anonymous}::ToString(Pinetime::Controllers::MotionController::DeviceTypes)':
C:/Git/PineTime/src/displayapp/screens/SystemInfo.cpp:18:12: warning: enumeration value 'Unknown' not handled in switch [-Wswitch]
   18 |     switch (deviceType) {
      |            ^
[297/527] Building C object src/CMakeFiles/lvgl.dir/libs/lvgl/src/lv_hal/lv_hal_tick.c.obj
In file included from c:\git\pinetime\src\libs\lvgl\src\lv_conf_internal.h:39,
                 from C:/Git/PineTime/src/libs/lvgl/src/lv_hal/lv_hal_tick.h:16,
                 from C:/Git/PineTime/src/libs/lvgl/src/lv_hal/lv_hal_tick.c:9:
C:/Git/PineTime/src/libs/lvgl/src/lv_hal/lv_hal_tick.c: In function 'lv_tick_get':
C:/Git/PineTime/src/libs/lv_conf.h:298:39: warning: implicit declaration of function 'xTaskGetTickCount' [-Wimplicit-function-declaration]
  298 | #define LV_TICK_CUSTOM_SYS_TIME_EXPR (xTaskGetTickCount()) /*Expression evaluating to current system time in ms*/
      |                                       ^~~~~~~~~~~~~~~~~
C:/Git/PineTime/src/libs/lvgl/src/lv_hal/lv_hal_tick.c:73:12: note: in expansion of macro 'LV_TICK_CUSTOM_SYS_TIME_EXPR'
   73 |     return LV_TICK_CUSTOM_SYS_TIME_EXPR;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[318/527] Linking CXX executable src\pinetime-app-1.2.0.out
FAILED: src/pinetime-app-1.2.0.out 
cmd.exe /C "cd . && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-c++" --sysroot="C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin" -g -mthumb -mabi=aapcs -L C:/Git/nRF5_SDK_15.3.0_59ac345/modules/nrfx/mdk -TC:/Git/PineTime/gcc_nrf52.ld -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,--gc-sections --specs=nano.specs -lc -lnosys -lm -Wl,-Map=pinetime-app-1.2.0.map @CMakeFiles\pinetime-app.rsp -o src\pinetime-app-1.2.0.out -Wl,--out-implib,src\libpinetime-app-1.2.0.dll.a -Wl,--major-image-version,0,--minor-image-version,0  && cmd.exe /C "cd /D C:\Git\PineTime\build\src && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-size" pinetime-app-1.2.0.out && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objcopy" -O binary pinetime-app-1.2.0.out pinetime-app-1.2.0.bin && "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-objcopy" -O ihex pinetime-app-1.2.0.out pinetime-app-1.2.0.hex""
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
[... cut ...]
[323/527] Building CXX object src/CMakeFiles/pinetime-recovery.dir/components/ble/ImmediateAlertService.cpp.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command Ninja" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

On a slightly off-topic note, I saw in pull request 505 ([link](https://github.com/JF002/InfiniTime/pull/505)) support for docker-like structure inside VSCode being developed - so maybe my issue is temporarily.

tlouwers, 2021-07-22:
For the linking issue I found that adding in 'tasks.json' the option:
     "-DCMAKE_SYSTEM_NAME=Generic",
make the error disappear.

Next I ran into the issue that 'python3' is not found. The python executable on windows is named python.exe and can be found (it was in my PATH), but pyrhon3 not. I made a copy of python.exe and renamed it to python3.exe - which worked fine.

My final tasks.json now looks like this:
`{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Configure - PineTime [Debug]",
            "type": "shell",
            "command": "CMake",
            "args": [
                "-B./build",
                "-GNinja",
                "-DCMAKE_BUILD_TYPE=Debug",
                "-DCMAKE_C_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe",
                "-DCMAKE_CXX_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe",
                "-DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY",
                "-DCMAKE_SYSTEM_NAME=Generic",
                "-DUSE_OPENOCD=1",
                "-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major",
                "-DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345",                
                "-DBUILD_DFU=1"                
            ],
            "problemMatcher": []
        },
        {
            "label": "Configure - PineTime [Release]",
            "type": "shell",
            "command": "CMake",
            "args": [
                "-B./build",
                "-GNinja",
                "-DCMAKE_BUILD_TYPE=Release",
                "-DCMAKE_C_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe",
                "-DCMAKE_CXX_COMPILER=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-g++.exe",
                "-DCMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY",
                "-DCMAKE_SYSTEM_NAME=Generic",
                "-DUSE_OPENOCD=1",
                "-DARM_NONE_EABI_TOOLCHAIN_PATH=C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major",
                "-DNRF5_SDK_PATH=C:/Git/nRF5_SDK_15.3.0_59ac345",                
                "-DBUILD_DFU=1"
            ],
            "problemMatcher": []
        },
        {
            "label": "Build - PineTime",
            "type": "shell",
            "options": {
                "cwd": "${workspaceFolder}/build"
            },
            "command": "Ninja",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
    ]
}`

Configuration works fine.
Compiling & linking works fine.
Python magic works fine.

The end result is a build folder with all expected output files.

JF002, 2021-07-22:
The definition of variables like `CMAKE_SYSTEM_NAME`, `CMAKE_TRY_COMPILE_TARGET_TYPE`, `CMAKE_CXX_COMPILER` and `CMAKE_C_COMPILER` is done in the toolchain configuration file `cmake-nRF5x/arm-gcc-toolchain.cmake`, it's strange it does not work as expected on Windows...

A little tip : Github has a formating feature to display log messages more nicely, you just need to wrap you logs with  \`\`\` (3 times ` and a line break). It'll make the logs easier to read ;)

closed 2021-10-02 by Avamander

lectrician1, 2021-10-03:
@Avamander I am too still not able to compile on Windows. Please do not close this.

Avamander, 2021-10-03:
@lectrician1 

> I am too still not able to compile on Windows. Please do not close this.

Open a separate issue where you describe what you're seeing and what your environment is like.