InfiniTimeOrg/InfiniTime #518: Music screen in some Android players don't show pause button

hubmartin, 2021-07-23:
Some android players seems like don't notify Infinitime over BLE so the player doesn't switch **Play** button in the UI to the **Pause** button. So I'm able to start playing but not to pause.

If I change code to send pause BLE command for example on the swipe down function, then the music pauses everytime. So it seems like just the Infinitime UI issue.
Hope that video below shows the problem in SoundCloud player.

https://photos.app.goo.gl/y97CDjg3RaziCgZc6

I just noticed, that when this issue occures that the disc animation in the top-right of the watch screen is not running. Seems like the `Refresh()` function is not called? I'm not familiar how this BLE service works.

In other apps and Podkicker it behaves properly. So it might be issue of the app, but I hope we can discuss here some ideas how to improve that or create some workaround. I could fix that, but I would like to hear your ideas too.

Some ideas:
- Swipe up/down changes next/prev and loc+/vol- buttons. So maybe swipe down will switch these controls and swipe up will send pause command?
- Double click on the play button or on the disc image will send pause command?
- Remove next/prev buttons and have just 2 fixed buttons play/pause. Volume will be controlled by swipe up/down, previous/next by swipe left/right? I have sometimes hard time to aim properly my finger to the middle button.
- your ideas.. ?

Thanks.

Avamander, 2021-07-23:
You haven't mentioned which InfiniTime version.

hubmartin, 2021-07-23:
Right, it is 1.3.0. But the same behavior had 1.2.0. I did not used older ones.

Avamander, 2021-07-23:
> Right, it is 1.3.0. But the same behavior had 1.2.0. I did not used older ones.

:+1: good to know. When reporting bugs this is important in quite a few cases.

> In other apps and Podkicker it behaves properly. So it might be issue of the app, but I hope we can discuss here some ideas how to improve that or create some workaround.

Should generally be reported to SoundCloud developers, they probably forgot to broadcast the intent that says the player actually started playing. Alternatively they decided to be special and have their own intents... for things that have standard ones :roll_eyes: 

Working around music player noncompliance is really an endless pit of misery, if you can't rely on anything it causes other issues. E.g. if you don't get confirmation that the music started playing but you change the UI, but this time it **actually** didn't start playing it's wrong in an another way and that does happen.

kennysc, 2021-09-10:
> Right, it is 1.3.0. But the same behavior had 1.2.0. I did not used older ones.

The same problem is occurs on 1.4.0 with spotify. But like @Avamander said, it might be a problem with the player.. Is there a way to confirm if this is the case or if it's coming from somewhere else?

hubmartin, 2021-09-10:
It is really issue with players. I've tested few of them and some worked fine. So I don't blame GadgetBridge or InfiniTime.

kennysc, 2021-09-10:
There was an update for Gadgetbridge (0.59.3) this morning and now it's working fine with spotify, though it might be a concidence.

Avamander, 2021-10-02:
If there are indicators that this is InfiniTime's problem then let's reopen.

closed 2021-10-02 by Avamander