InfiniTimeOrg/InfiniTime #525: How to produce merged-internal.hex

lynxD, 2021-07-25:
In release 1.2.0 merged-internal.hex file is provided. I can't find any documentation/script on procedure to produce it. Can anyone point me to it/write it down?

JF002, 2021-07-25:
The additional files in release 1.2.0 are the files I generated and sent to the factory for the production of the new batch of PineTimes:

 - merged-internal.hex contains the bootloader + infinitime (to be loaded in the internal memory of the nrf52)
 - spinor.bin contains the recovery firmware (to be flashed in the internal SPI flash memory).

I generated these files manually (but it would be nice if they were generated automatically by CMake) : 

- merged-internal.hex : 
  * Convert pinetime-mcuboot-app-image-x.y.z.bin to an hex file using objcopy (do not forget the option to specify the offset at 0x8000).
  * Merge this hex file with the one containing the [bootloader](https://github.com/JF002/pinetime-mcuboot-bootloader/releases/tag/1.0.0). For this, I used **merghex** from the [NRF CLI tools](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf_cltools%2FUG%2Fcltools%2Fnrf_mergehex.html).

- spinor.bin : 
  * Convert the file  pinetime-mcuboot-recovery-image-1.3.99.bin to hex using objcopy. The offset is 0x00.

lynxD, 2021-09-08:
I've successfully generated merged-internal.hex and spinor.bin. Thx!
The commands I've used:
merged-internal.hex:
```
 arm-none-eabi-objcopy --input-target=binary --output-target=ihex --change-section-address '*=0x8000' build/output/src/pinetime-mcuboot-app-image-1.3.0.bin build/output/src/pinetime-mcuboot-app-image-1.3.0.hex
path/to/nRF-cli/nRF-Command-Line-Tools_10_13_0_Linux64/mergehex/mergehex --merge build/output/src/pinetime-mcuboot-app-image-1.3.0.hex ~/tmp/bootloader.hex --output build/merged-internal.hex
```
spinor.bin:
```
arm-none-eabi-objcopy --input-target=binary --output-target=ihex build/output/src/pinetime-mcuboot-recovery-image-1.3.0.bin build/output/src/spinor.bin
```

I can add that to build logic. I just need to know how do You want it go about getting bootloader? Add repo as a submodule(specific commit) and build it? Expect a file to be placed somewhere and not build by default? Add a golden binary to git?

closed 2021-10-02 by Avamander