InfiniTimeOrg/InfiniTime #527: Bluetooth Connection Lost Randomly, Requires Reset

atomdmac, 2021-07-25:
My phone will randomly disconnect from my `PineTime` and will not connect or advertise itself until it is reset.

This seems to happen at random intervals and during a time when I am not actively using the watch.  The shortest interval so far as been ~3 hrs and the longest has been ~2 days.

**The strangest bit**: I'm using `Gadget Bridge` as my companion app but when this issue occurs, I cannot re-connect until after I've connected + disconnected via `nrfConnect`.

The full "reset procedure" is:
1. Reset the `PineTime`
2. Open `nrfConnect` and connect to the `PineTime`
3. Disconnect from the `PineTime` via `nrfConnect` and close `nrfConnect`
4. Open `Gadget Bridge` and connect to the `PineTime`

Has anyone else experienced this problem?

**Phone:** Pixel 2
**InfiniTime Version**: `1.2.0` and `1.3.0`

Avamander, 2021-07-25:
Duplicate of https://github.com/JF002/InfiniTime/issues/302

JF002, 2021-07-30:
Several users reported that kind of issue (me included). In my case, I have to reset the watch ~ once a day. I do it in the morning so that BLE connectivity works the whole day.
You can have a look at #207, where we try to analyze this issue a bit further.

duckunix, 2021-07-31:
> Duplicate of #302
I would say it is not, as I experience my pinetime just dropping off with the phone in my pocket and the watch on my arm hanging around my house.  Or when I leave my phone on the desk and I go to the other end of the house.

demure, 2021-08-11:
With the 1.3.0 firmware and android+gadgetbridge, I can recreate this issue when the pinetime is taken out of bluetooth range from the phone. It seems as though the pinetime ends up in some kind of fail condition where it no longer attempts (or at least fails at) reconnecting.

justinesmithies, 2021-08-12:
Just got my PineTime today and updated to the latest. I too am using gadgetbridge and it seems to lose its connection. Resetting restores connectivity.

bearclaws8, 2021-08-13:
Might it be that this issue and #302 are connected but due to two different issues, one that GB is a flaky and the other where PineTime stops broadcasting? Once PineTime is not seen in my phone's Bluetooth connections the only option I have found is to reboot the PineTime, sometimes more than once.

Is it possible to add into the settings the ability to turn Bluetooth on/off and force the service to restart without restarting the entire device?  On/off may be nice for those looking to maximize battery and security and restarting the BLE broadcast may help with debugging.

shymega, 2021-08-29:
Yeah, got my PineTime yesterday, updated to the latest - doing the 8-second button hold and rebooting fixes it. I also updated the bootloader and recovery, but I don't think that's related. I use a Cosmo Communicator, and I disabled DuraSpeed on it, just in case *that* was the issue - same behaviour. At this stage, I suspect it is either GB or the firmware. I'm leaning towards the firmware.. it's almost like the BLE connection stops broadcasting, as my dev machine can't 'see' the PineTime at all.

demure, 2021-08-30:
As an odd -- and not yet replicated -- data point, my watch managed to reconnect on it's own over a three to four day period.
I had left the watch on the charger friday, and my phone (currently on gadgetbridge 0.59.1) went out of bluetooth range of the watch a handful of times starting on friday. As of this monday morning, my watch was connected with an uptime of ~4d19h, and gb's power graph suggested the watch reconnected around the ~4d uptime mark.

JF002, 2021-09-30:
InfiniTime 1.5 and 1.6 brought many improvements the BLE connectivity. While I doubt those changes fix **all** connectivity issues, they certainly fix most of them. I think this one (BLE that stops working after a day) is one of them. I'll close this issue, but feel free to open a new one if you encounter another BLE issue on InfiniTime 1.6 (and if it doesn't exist yet).

closed 2021-09-30 by JF002