InfiniTimeOrg/InfiniTime #528: Screen sometimes doesn't wake up with 1.3.0

sirnewton01, 2021-07-26:
I updated to the 1.3.0 firmware and it seems that sometimes the screen won't wake up. Putting it on the charger usually makes it responsive again. It keeps the time and so I don't expect that it locks up or reboots.

I didn't experience this with the 1.2.0 firmware, so it seems like it could be a regression.

jonvmey, 2021-07-26:
Which wakeup method(s) does it fail to respond to?

sirnewton01, 2021-07-26:
When it happens (twice now), it doesn't respond to the physical button nor the double-tap of the screen, which is my configured wakeup gesture. I haven't rebooted it yet, since I haven't validated the new firmware.

Riksu9000, 2021-07-26:
Sounds like this might be related to #512. There people have also mentioned difficulty getting the device to wake up.

sirnewton01, 2021-07-26:
I'll try charging it fully to see if it happens again. It's weird that I never saw this with the 1.2 firmware.

sirnewton01, 2021-07-27:
Ok, so far it seems that after a full charge it isn't happening again. So, perhaps the battery was lower than reported on the watch face.

closed 2021-07-27 by sirnewton01