InfiniTimeOrg/InfiniTime #536: Steps not counted with BMA425

TheB3st3st, 2021-07-29:
The steps stay at 0, but the m tab shows accurate data. Issue persists with all versions.

JF002, 2021-07-30:
It looks like the motion sensor failed to initialize properly. From your description, it looks like the accelerometer works but not the step counter (which is part of the accelerometer).

You can try to reset the watch (long push on the button) and see if it improves the situation.
Next, you can also wait until the battery is completely drained before charging the watch again. This way, you'll ensure a hardware reset is applied.

atomdmac, 2021-08-08:
@TheB3st3st Is the "0 step count" displayed on the watch face?  Mine increments as expected but I have noticed that there is a "steps" field on the 3rd page of the "About" section of the settings that always reads "0".  Not sure if that's the same issue that you're describing, though.

1989gironimo, 2021-09-27:
I have a similar issue now, but I suspect a defect of the sensor.
Will report back after a hardware reset.

Avamander, 2021-10-02:
Sounds like a hardware issue, it works for everyone else in the same setup.

closed 2021-10-02 by Avamander

1989gironimo, 2021-10-02:
Can confirm, even after a hardware reset, steps arent counted and the motion app doesn't show any activity.