InfiniTimeOrg/InfiniTime #540: Watchface with bigger font for Date and Health/Fitness Data

Hensch-S, 2021-07-31:
I wanted to give a Pine time to my father as a gift but I'm afraid of the fonts on the Display being too small to view without reading glasses. Would it be possible to change the digital Watchface or create a new one. The time on digital watchface is pretty readable but I think the date, HRM, step count and status symbols have to be bigger.

Thanks in advance.

Dperrigo18, 2021-08-02:
The only problem I could think of is that certain things could clip into other since the spacing on all of the screens are different.
If this is just about reorganizing the main watch face in general (making icons bigger and moving things around) there is an open issue about having the ability to have custom watch faces. #289

Riksu9000, 2023-02-02:
I want to look into this issue to assess the accessibility of InfiniTime. Anyone reading this, please share your experiences. Is the font too small? How much larger should it be? Here are two builds based on 1.11.0 with larger font sizes. Please test them and let me know what you think.

[infinitime-font-large.zip](https://github.com/InfiniTimeOrg/InfiniTime/files/10566738/infinitime-font-large.zip)
[infinitime-font-larger.zip](https://github.com/InfiniTimeOrg/InfiniTime/files/10566740/infinitime-font-larger.zip)

JF002, 2023-02-12:
This is a feedback I received at FOSDEM : older eyes have a hard time reading the smaller font (date on the digital watchface, notification text, texts in the settings,...).

I'm not involved in this issue (yet), so it's not easy to determine whether the font and layout are easily readable by anyone or not. There are also other challenges with bigger fonts : the display is still quite small, and bigger fonts mean more flash space needed to store them.

tomechio, 2023-03-05:
Hi,

For some reason I'm not able to download the file with larger font, but the version with large font already feels better for my middle age eyes :)

agittins, 2023-03-07:
An anecdotal datapoint - at 45 my near vision is getting further away than my arm can. If I don't have reading glasses on, the date on the digital face is pretty borderline for me, and is usually "difficult" to read. The icons in the swipe-up menu are frustratingly small, given they occupy a very small area of the buttons (this is on 1.11.0). The text in the menus and the "firmware" info screen though are pretty readable, being bold and bright.

My preference would be for nudging up the sizes a bit, but by the same token I should probably be wearing my glasses 🤣

ajack2001my, 2023-03-08:
Maybe a systemwide flag for FontSize=(small,standard,large) and let each app dev decide their own app font pitch with guidelines like normal=x points, small=90% of standard, large=120% of standard.

Riksu9000, 2023-03-08:
Thank you for your feedback so far.

The fact that the firmware info screen is more readable is interesting. As an alternative to increasing the font size, maybe we can consider increasing the contrast of small text, as increasing the font size has drawbacks regarding how much text can be displayed. Perhaps we can also experiment with using another font? #955

In the next release the buttons in the settings list will have a dark gray background, increasing contrast. Can we resolve this issue by making changes like this? What else needs changing? The app screen can certainly be improved by increasing the icon size and contrast. We can make the digital watch face's date text slightly brighter if necessary.

Including multiple similar font sizes is unfortunately not an option, because each font uses a lot of memory.

[image]

tomechio, 2023-03-08:
For me the  biggest issue are the notifications. With buttons etc. I already remember where everyhing is, so it doesn't matter if the texts of the buttons are little bit fuzzy. But the notifications have always a different content so I need to be able to read them clearly.

Riksu9000, 2023-03-11:
@tomechio Are you using version 1.11, where the notification colors were changed? Is it still an issue?

tomechio, 2023-03-13:
@Riksu9000 Yes, these colors are better than the old ones (yellow text on a grey background)

I think that I will build my own version with bigger font 20px -> 24px and at the same time replace the cyrillic alphabets with letters Å Ä Ö that go better with Finnish language (so I don't have to use the transliteration in Gadgetbridge)

yuval-herman, 2023-03-23:
I have read through the comments and thought I could just draw your attention to this: https://en.wikipedia.org/wiki/Pixel-art_scaling_algorithms#EPX%2FScale2%C3%97%2FAdvMAME2%C3%97

I have no opinion on the matter in general, but perhaps this could solve the issue of storing bigger fonts.

Saarsk, 2023-05-21:
> There are also other challenges with bigger fonts : the display is still quite small, and bigger fonts mean more flash space needed to store them.

Just like the issue with it not being able to show "æ, ø, å" and other local characters. Personally, I think it should be configurable. Perhaps be module-based or configurable pre-compile. I really hope you won't try to go for a one-size-fits-all-solution, as this never really solves the issue but instead just makes it less good for everyone as such 1-for-all solutions tries to find the average of all preferences. :)

How feasable is it to be able to configure it pre-compile or modify modules post-compile? I imagine either command line parameters or in a perfect world a visual configurator (but this requires a lot of development, so I want to make it clear I'm not expecting any of this).

In my instance I'd enable additional regional characters, small fonts, remove some apps, make room for others including additional watchfaces (and if I were to dream also choose color theme and font type and gestures).

I know this is a theoretical sky is the limit sort of thing where the practical reality is different due to time, resources and developers. So my bottom line point is, I think the small font should be default but I would also be happy to see a version (or configuration of sorts) for seniors or others in need of enlarged fonts.

agittins, 2023-05-22:
> Short conclusion: ... customisation is the answer. :) 

> (but this requires a lot of development, so I want to make it clear I'm not expecting that).

Completely agree. I think others would be happy to step in and help build tools for customisation, but there is currently no "path" for that which i am aware of for centrally defining which aspects or settings to build and i haven't seen anything in the roadmap etc except for a preference against ifdefs and valuing a default experience over personalisation.

No piece of tech i own is in constant contact with my body more than my pinetime. It's endlessly frustrating to me that it's so nearly super useful, but just a bit further from it than i can get on my own.

minacode, 2023-05-22:
I would like to add that smaller font sizes and more content per screen seem to be a "nice to have" feature, whereas larger font sizes are more necessary for some people to use the watch at all.