InfiniTimeOrg/InfiniTime #545: Integration with Google Fit and Apple Health

apcameron, 2021-08-02:
It would be great if the step tracking and heart rate information could be integrated or imported to Google Fit and Apple Health.

hatmajster, 2021-08-02:
As this is very free source project, probably there won't be many folks interested in doing this as no one is actually using those apps. But it would be nice feature. Note that GNU Health integration is in preparation

Avamander, 2021-08-02:
This is not really an InfiniTime thing to do. It's up to your choice of companion app.

ajack2001my, 2021-08-16:
I believe if the relevant health API hooks are there, it can be done in the Gadgetbridge app to sync to Google Fit. Is my assumption correct? If so, who here can do this? :)

Avamander, 2021-08-16:
> I believe if the relevant health API hooks are there

Step count at the time of writing is yet unavailable from InfiniTime's side.

> it can be done in the Gadgetbridge app to sync to Google Fit.

Can, but Gadgetbridge devs won't because they don't have phones with Google. If Google Fit API/bindings are with an open license, and someone else would implement it, they might merge. If someone here is interested in doing so, do the research and ask them first.

Avamander, 2021-10-02:
Companion-side task mostly.

closed 2021-10-02 by Avamander